espressif/esp-adf #620: stop/resume the audio pipeline crash

iloop2020, 2021-05-18:
Hi ,

I found if we stop/resume the audio pipeline too fast, ESP32 will crash at audio_pipeline_stop(), I suspect the resume() is not completely ready and call stop() will crash.

1. audio_pipeline_stop()
2. audio_pipeline_wait_for_stop()
3. audio_pipeline_resume()
4. Go to 1.

Please help.

Thank you

HengYongChao, 2021-05-19:
Hi @iloop2020 

I don’t recommend that you use it in this order.

Generally,  we use pipeline start and  stop in pairs,  pipeline pause and resume as the same way.

When you are already stop a pipeline then continue to resume the pipeline, maybe got an issue.

iloop2020, 2021-05-19:
Hi @HengYongChao 

Thank you for your support, it is good to know more in detail.

1. Can you list out the API pairs that should be called?
2. What is different between audio_pipeline_terminate() and audio_pipeline_stop() ?

Thank you.

iloop2020, 2021-05-19:
Hi @HengYongChao 

The document shows the stop() can be use with resume(), please check.
[image]

Thank you

HengYongChao, 2021-05-20:
Hi @iloop2020  

Sorry, I thought you were using the `audio_pipeline_terminate` function, resume is more like restart sometime. 
In the above case, you can surely use resume.

Please post the crash log here.

iloop2020, 2021-05-23:
Hi @iloop2020 

Can you share the API call flow/produce, to switch between Bluetooth A2DP and SDCard MP3 file playback. ?

Thank you

iloop2020, 2021-05-29:
Hi @HengYongChao 

1. What is the different between audio_pipeline_pause() and audio_pipeline_stop() ?
2. And the relation between audio_pipeline_start() and audio_pipeline_resume() ?

Thank you