espressif/esp-adf #621: VOIP deep sleep issue

Patsjemoe, 2021-05-18:
Hi, I added esp_deep_sleep_start(); to the  end of "case SIP_EVENT_AUDIO_SESSION_END:" ,  with the idea to put the device to sleep when my audio session is ended.
It seems to unlink my pipeline immediately, not waiting untill the end of the audio session,  so do not get voice.
Any idea of what I overlook ? 

I (14127) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_END
W (14127) AUDIO_ELEMENT: IN-[filter] AEL_IO_ABORT
W (14127) AUDIO_ELEMENT: IN-[sip_dec] AEL_IO_ABORT
I (14147) SIP_RTP: send task stopped
W (14187) AUDIO_PIPELINE: There are no listener registered
I (14187) AUDIO_PIPELINE: audio_pipeline_unlinked
W (14187) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (14197) AUDIO_ELEMENT: [sip_dec] Element has not create when AUDIO_ELEMENT_TER
MINATE
W (14207) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERM
INATE
W (14217) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (14227) AUDIO_ELEMENT: IN-[sip_enc] AEL_IO_ABORT
W (14247) AUDIO_PIPELINE: There are no listener registered
I (14247) AUDIO_PIPELINE: audio_pipeline_unlinked
W (14247) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (14257) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERM
INATE
W (14267) AUDIO_ELEMENT: [sip_enc] Element has not create when AUDIO_ELEMENT_TER
MINATE
W (14277) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (14287) AUDIO_ELEMENT: [algo] Element has not create when AUDIO_ELEMENT_TERMIN
ATE

thanks in advance
best regards
Ludo

HengYongChao, 2021-05-19:
Hi @Patsjemoe 

When you call `esp_deep_sleep_start()` , it means the CPUs were shut down, and lose all data. 
I think you should read the API comment before using it.

Patsjemoe, 2021-05-19:
@ HengYongChao :
This is clear to me, but I want to put the device to sleep after the call
is finished/ hangup and startup the board and make a call when io33 is
pulled high.
If I use deep sleep after my audio session is ended, I should be able to
have my call and then go to sleep.
It should not go asleep before the end of my call.
Where then in the Voip session would you use the deep sleep start ?
Regards
Ludo

Patsjemoe, 2021-05-19:
@ HengYongChao :
I would like to add that when I put deep sleep in  SIP_EVENT_HANGUP, 
        case SIP_EVENT_HANGUP:
            ESP_LOGI(TAG, "SIP_EVENT_HANGUP");
	    esp_deep_sleep_start();
            break;
     Then I can make my call, and when I finish my call/hangup , it goes to sleep 
     when my phone rings, and pick it up, I get following SIP message 
     I (36677) SIP:
                     SIP/2.0 200 OK

But, when I put it at the end of  SIP_EVENT_AUDIO_SESSION_END, which means after I should have made a call and then end the audio session, I do not get into my audio session and below pipelines are stopped and my session stops before I have it....this seems to be strange.
       case SIP_EVENT_AUDIO_SESSION_END:
            ESP_LOGI(TAG, "SIP_EVENT_AUDIO_SESSION_END");
            audio_pipeline_stop(player);
            audio_pipeline_wait_for_stop(player);
            audio_pipeline_deinit(player);
            audio_pipeline_stop(recorder);
            audio_pipeline_wait_for_stop(recorder);
            audio_pipeline_deinit(recorder);
	   esp_deep_sleep_start();
            break;
When my phone rings, and I pick up, I get following SIP message : 
I (21477) SIP_RTP: send task stopped <<<<<<<---------------------------
I (21487) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_END

Note : what you said was : you put it to sleep, so obviously everything stops....but as deep sleep start is after the audio pipelines should be stopped, deep sleep cannot cause the pipelines to stop before sleep is called...??
Maybe I overlook something, but it seems strange that the audio session stops before I had it.

regards
Ludo

Patsjemoe, 2021-05-22:
@ HengYongChao :
One more strange observation that I did not notice before : 
It seems that the program first has a SIP_EVENT_AUDIO_SESSION_END before starting  a SIP_EVENT_AUDIO_SESSION_BEGIN, which makes that in case I put deep sleep after SIP_EVENT_AUDIO_SESSION_END, it will not reach SIP_EVENT_AUDIO_SESSION_BEGIN anymore. Why it does first a SIP_EVENT_AUDIO_SESSION_END.....I do not know ? is this to be sure there is no more session active ? It seems to have the same behaviour without deep sleep.
regards
Ludo

******************************************************************************
I (22753) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_END
W (22753) AUDIO_ELEMENT: OUT-[sip_dec] AEL_IO_ABORT
W (22753) AUDIO_ELEMENT: OUT-[filter] AEL_IO_ABORT
I (22773) SIP_RTP: send task stopped
W (22813) AUDIO_PIPELINE: There are no listener registered
I (22813) AUDIO_PIPELINE: audio_pipeline_unlinked
W (22813) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22823) AUDIO_ELEMENT: [sip_dec] Element has not create when AUDIO_ELEMENT_TER
MINATE
W (22833) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERM
INATE
W (22833) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22863) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (22863) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (22863) AUDIO_PIPELINE: There are no listener registered
I (22873) AUDIO_PIPELINE: audio_pipeline_unlinked
W (22873) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22883) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERM
INATE
W (22893) AUDIO_ELEMENT: [sip_enc] Element has not create when AUDIO_ELEMENT_TER
MINATE
W (22903) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22913) AUDIO_ELEMENT: [algo] Element has not create when AUDIO_ELEMENT_TERMIN
ATE
I (22923) SIP_RTP: receive task stopped
I (22923) SIP_RTP: RTP stopped
W (22923) SIP: CHANGE STATE FROM 72, TO 96, :func: _sip_uac_process_res_ok:1126
I (22923) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_BEGIN

Patsjemoe, 2021-05-25:
@ HengYongChao  :
 it is strange, because this behaviour is not happening with esp32 audiokit, and I  think I did not see it with LyraT.
regards
Ludo

Patsjemoe, 2021-05-27:
> @ HengYongChao :
> One more strange observation that I did not notice before :
> It seems that the program first has a SIP_EVENT_AUDIO_SESSION_END before starting a SIP_EVENT_AUDIO_SESSION_BEGIN, which [...]

@ HengYongChao :can you confirm ifyou have the same behaviour?  Was the SIP library changed?

Patsjemoe, 2021-06-02:
Hi,
I will close this issue, as it is no longer a deep sleep issue.
Will make a new one on VOIP SIP behaviour problem.
regards
Ludo

closed 2021-06-02 by Patsjemoe