espressif/esp-adf #625: Stack overflow error on multi-room advanced example

convertl, 2021-05-21:
Hello,

I would like to try the multi room audio example with 3 LyraT mini v1.2 boards.
I applied the I2S patch and compiled with IDF 4.2.1, but I get a stack overflow error each time I press Play or Rec key 👍 

`----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
|                       ESP_AUDIO-v1.7.0-037bef3-09be8fe                     |
|                     Compile date: May  8 2021-17:38:00                     |
------------------------------------------------------------------------------
I (5429) ESP_AUDIO_CTRL: Func:media_ctrl_create, Line:349, MEM Total:4296060 Bytes, Inter:234944 Bytes, Dram:207752 Bytes

I (5441) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (5463) MRM_EXAMPLE: [ 3 ] Create and start input key service
I (5463) ESP_DECODER: esp_decoder_init, stack size is 30720
I (5465) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (5487) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (5490) LYRAT_MINI_V1_1: I2S0, MCLK output by GPIO0
I (5497) ESP_AUDIO_CTRL: Enter play procedure, src:0
I (5502) ESP_AUDIO_CTRL: Play procedure, URL is ok, src:0
I (5508) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffd3878
I (5515) ESP_AUDIO_TASK: It's a decoder
I (5520) ESP_AUDIO_TASK: 1.CUR IN:[IN_raw],CODEC:[DEC_auto],RESAMPLE:[48000],OUT:[OUT_iis],rate:0,ch:0,pos:0
I (5531) ESP_AUDIO_TASK: 2.Handles,IN:0x3f8120bc,CODEC:0x3f8123b4,FILTER:0x3f8126c8,OUT:0x3f812574
I (5540) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (5545) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (5551) AUDIO_PIPELINE: link el->rb, el:0x3f8120bc, tag:IN_raw, rb:0x3f8129d0
I (5559) AUDIO_PIPELINE: link el->rb, el:0x3f8123b4, tag:DEC_auto, rb:0x3f815210
I (5567) AUDIO_PIPELINE: link el->rb, el:0x3f8126c8, tag:Audio_forge, rb:0x3f817250
I (5575) ESP_AUDIO_TASK: 3. Previous starting...
I (5580) AUDIO_ELEMENT: [IN_raw-0x3f8120bc] Element task created
W (5587) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (5595) AUDIO_ELEMENT: [DEC_auto-0x3f8123b4] Element task created
I (5602) AUDIO_ELEMENT: [DEC_auto] AEL_MSG_CMD_RESUME,state:1
I (5609) ESP_AUDIO_TASK: No Blocking play with decoder
I (5614) ESP_AUDIO_TASK: Recv Element[IN_raw-0x3f8120bc] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (5625) ESP_AUDIO_TASK: Recv Element[IN_raw-0x3f8120bc] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (5636) ESP_DECODER: Ready to do audio type check, pos:0, (line 104)
I (5644) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:770, MEM Total:4224508 Bytes, Inter:189628 Bytes, Dram:162436 Bytes

I (5655) ESP_AUDIO_CTRL: Exit play procedure, ret:0
W (5663) MRM_MULTICAST: creat addr 239.255.255.252 port 1900
W (5667) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (5675) MRM_CLIENT: Slave start searching...
I (144565) MRM_EXAMPLE: [ * ] [Play] input key event
I (144680) MRM_CLIENT: slave client task stoped
W (144681) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1

***ERROR*** A stack overflow in task input_key_servi has been detected.

Backtrace:0x4008b661:0x3ffd4c60 0x4008bc71:0x3ffd4c80 0x4008bf2d:0x3ffd4ca0 0x4008cb79:0x3ffd4d20 0x4008c020:0x3ffd4d60 0x4008bfd6:0x3ff00020 |<-CORRUPTED       
0x4008b661: panic_abort at C:/Users/user/esp.42/esp-idf/components/esp_system/panic.c:341

0x4008bc71: esp_system_abort at C:/Users/user/esp.42/esp-idf/components/esp_system/system_api.c:106

0x4008bf2d: vApplicationStackOverflowHook at C:/Users/user/esp.42/esp-idf/components/freertos/xtensa/port.c:436

0x4008cb79: vTaskSwitchContext at C:/Users/user/esp.42/esp-idf/components/freertos/tasks.c:2771

0x4008c020: _frxt_dispatch at C:/Users/user/esp.42/esp-idf/components/freertos/xtensa/portasm.S:431

0x4008bfd6: _frxt_int_exit at C:/Users/user/esp.42/esp-idf/components/freertos/xtensa/portasm.S:231`

Is there any specific configuration for LyraT mini v1.2 to get the PSRAM issue workaround working for this example ?

Thanks

HengYongChao, 2021-05-22:
HI @convertl 

Have you applied the patch [idf_v3.3_freertos.patch](https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v3.3_freertos.patch)  yet? 

I see the warning:  `W (144681) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1`

convertl, 2021-05-25:
Hello @HengYongChao,

Thanks for the reply and the suggestions.
Applying the content of the idf_v3.3_freertos.patch on my idf 4.2 freertos, and increasing the input_key_service stack to 4*1024 instead of 3*1024 made the master card working (I can get sound out of it, and I see the MRM CLIENT Send Sync frames sending.

However, from the slave side, it is not working (I don't think it is due to stack anymore).
I documented the issue on [issue 627](https://github.com/espressif/esp-adf/issues/627)

binartech, 2021-12-23:
> HI @convertl
> 
> Have you applied the patch [idf_v3.3_freertos.patch](https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v3.3_freertos.patch) yet?
> 
> I see the warning: `W (144681) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1`
> 
> It seems that the chip version of your development board is not [ESP32_ECO_V3](https://www.espressif.com/sites/default/files/documentation/ESP32_ECO_V3_User_Guide__EN.pdf).
> 
> I suggest you increase the ·input_key_service task· stack first, and See if the problem can be solved.

starting to Work on this, may be my question is too basic, but where you change the input_key_service task stack ? This is happening with the Lyra-T 4.3 board and idf-4.2. The freertos patch doesn't work for latest idf I tested, the 4.3.2, which also seems not supporting the multi-room example.
Also I get glitches in Audio. The error related to the Input Key service happens after restarting the board with the firmware flashed.
The first time it runs OK. Even reflashing doesn't solves the issue. Erasing the flash and than reflashing make it works but just once.
Thanks

jason-mao, 2022-02-07:
@binartech  
> starting to Work on this, may be my question is too basic, but where you change the input_key_service task stack ? 

Please increase the task_stack on 
https://github.com/espressif/esp-adf/blob/c5ef39d7c3ae3e3f4fb997b81dfc5d14b36ba299/components/input_key_service/include/input_key_service.h#L88

closed 2022-02-07 by jason-mao