espressif/esp-adf #626: esp-adf-v2.3 VOIP issue on lyraT

Patsjemoe, 2021-05-25:
Hi,
I installed adf-v2.3 , and started VOIP example on LyraT  to test if this has the same strange behaviour as LyraT mini.
Unfortunately, there is an error that fails to get voice : It seems to initialize other parts like dec_mp3 that I think should not be there.

I (3051) VOIP_EXAMPLE: SIP_EVENT_REGISTERED
I (3051) ESP_AUDIO_CTRL: Enter play procedure, src:0
I (3061) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffb5784
I (3071) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:713, MEM Total:4361020 Bytes, Inter:276996 Bytes
, Dram:217900 Bytes

I (3071) ESP_AUDIO_TASK: It's a decoder
I (3081) ESP_AUDIO_TASK: 1.CUR IN:[IN_flash],CODEC:[DEC_mp3],RESAMPLE:[16000],OUT:[OUT_iis],rat
e:0, ch:0
I (3091) ESP_AUDIO_TASK: 2.Handles,IN:0x3f8061cc,CODEC:0x3f806364,FILTER:0x3f80789c,OUT:0x3f806
4f4
I (3101) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (3111) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (3111) AUDIO_PIPELINE: link el->rb, el:0x3f8061cc, tag:IN_flash, rb:0x3f807a5c
I (3121) AUDIO_PIPELINE: link el->rb, el:0x3f806364, tag:DEC_mp3, rb:0x3f807a98
I (3131) AUDIO_PIPELINE: link el->rb, el:0x3f80789c, tag:resample, rb:0x3f807ad4
I (3141) ESP_AUDIO_TASK: 3. Previous starting...
I (3141) AUDIO_ELEMENT: [IN_flash-0x3f8061cc] Element task created
I (3151) AUDIO_ELEMENT: [IN_flash] AEL_MSG_CMD_RESUME,state:1
I (3151) flashPartition: 146: label:flash_tone
I (3151) AUDIO_THREAD: The DEC_mp3 task allocate stack on external memory
I (3161) TONE_STREAM: header tag C442, format -577764970
I (3171) AUDIO_ELEMENT: [DEC_mp3-0x3f806364] Element task created
E (3171) AUDIO_ELEMENT: [IN_flash] AEL_STATUS_ERROR_OPEN,-1
I (3181) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (3191) MP3_DECODER: MP3 opened
E (3201) ESP_AUDIO_TASK: Got element[IN_flash] error:ESP_ERR_AUDIO_OPEN, stop pipeline[0]
W (3201) AUDIO_ELEMENT: IN-[DEC_mp3] AEL_IO_ABORT
W (3211) AUDIO_PIPELINE: Without stop, st:1
E (3211) MP3_DECODER: failed to read audio data (line 117)
W (3221) AUDIO_PIPELINE: Without wait stop, st:1
E (3221) AUDIO_ELEMENT: [DEC_mp3] AEL_STATUS_ERROR_OPEN,-1
W (3231) AUDIO_ELEMENT: [IN_flash] Element already stopped
I (3231) MP3_DECODER: Closed
W (3241) AUDIO_ELEMENT: [DEC_mp3] Element already stopped
W (3201) AUDIO_ELEMENT: [IN_flash] AEL_MSG_CMD_ERROR,6
W (3251) ESP_AUDIO_TASK: Destroy the old pipeline
W (3261) AUDIO_ELEMENT: [OUT_iis] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3271) AUDIO_ELEMENT: [resample] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3281) ESP_AUDIO_TASK: The old pipeline destroyed
I (3281) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffb5784
I (3291) ESP_AUDIO_TASK: Received last pos: 0 bytes
I (3301) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:ERROR, 0
I (3301) ESP_AUDIO_CTRL: Exit play procedure, ret:81100
W (3311) SIP: CHANGE STATE FROM 1, TO 2, :func: sip_register:1592

then try to make a call, but get
 W (6191) SIP: CHANGE STATE FROM 2, TO 64, :func: esp_sip_uac_invite:1954 and call is not made.
if I push again play, I get I (87791) VOIP_EXAMPLE: [ * ] [Play] input key event

This behaviour was not there in rev 2.1 VOIP .

regards
Ludo

Patsjemoe, 2021-06-14:
Hi,
Can someone please run the Voip example and confirm if my observation is confirmed? 
Is adfv2.3 released, or still in beta? 
Regards
Ludo

Patsjemoe, 2021-07-04:
@jason-mao : 
SInce may25th, I am trying to understand what is happening with VOIP in ADF v2.3, but no answer.
Please confirm how to solve the errors  ?

/E (3123) TONE_PARTITION: Not flash tone partition

W (3123) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
E (3133) AUDIO_ELEMENT: [IN_flash] AEL_STATUS_ERROR_OPEN,-1

E (3183) TONE_PARTITION: C:/msys32/home/ludok/esp/esp-adf-v2.3/components/tone_partition/tone_partition.c:2
04 (tone_partition_deinit): Got NULL Pointer

E (3213) ESP_AUDIO_TASK: Got element[IN_flash] error:ESP_ERR_AUDIO_OPEN, stop pipeline[0]
E (3213) MP3_DECODER: failed to read audio data (line 117)
E (3223) AUDIO_ELEMENT: [IN_flash] Element already stopped
W (3233) AUDIO_ELEMENT: [DEC_mp3] AEL_IO_ABORT, -3
I (3243) MP3_DECODER: Closed
E (3253) AUDIO_ELEMENT: [IN_flash] Element already stopped
E (3253) AUDIO_ELEMENT: [DEC_mp3] Element already stopped

Note : why is this "tone" added to VOIP ?

regards

Patsjemoe, 2021-07-04:
[image]
@jason-mao  : please confirm how to solve these tone problems ?
regards
Ludo

ahhfzhang, 2021-07-05:
Hi @Patsjemoe,

I think you should add a tone partition on you csv like this https://github.com/espressif/esp-adf/blob/v2.3/examples/advanced_examples/voip/partitions_voip_example.csv#L6

and download the flash tone by Readme.

BR.

Patsjemoe, 2021-07-05:
Hi,
thank you, this solved the problem.
Regards
Ludo

closed 2021-07-05 by Patsjemoe