espressif/esp-adf #627: ADF Multi room advanced example : MRM_Multicast error on slave

convertl, 2021-05-25:
Hi,

After having (hopefully) fixed the stack error problem I had on this example, using LyarT-Mini v1.2 board (see [Issue 625: Stack overflow error on multi-room advanced example (AUD-3034)](https://github.com/espressif/esp-adf/issues/625)), I can play a remote mp3 file on the master, but on the slave, I get _MRM_Multicast error_ messages :

`
I (5654) ESP_AUDIO_CTRL: Exit play procedure, ret:0

W (5662) MRM_MULTICAST: creat addr 239.255.255.252 port 1900
W (5666) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (5674) MRM_CLIENT: Slave start searching...
E (6282) MRM_MULTICAST: select error, sock = 1936876886
E (6282) MRM_MULTICAST: select error, sock = 1936876886
E (6283) MRM_MULTICAST: select error, sock = 1936876886
E (6289) MRM_MULTICAST: select error, sock = 1936876886
E (6294) MRM_MULTICAST: select error, sock = 1936876886
`

Could you please help point out why there are such errors on the slave?

HengYongChao, 2021-06-08:
@ahhfzhang  Please take a look at this question.

ahhfzhang, 2021-06-21:
Hi @convertl ,

Due to  https://github.com/espressif/esp-adf/issues/625, I think maybe you should use another development board with the ESP32_ECO_V3 chip. 

by the way, MRM master and slave client are on the same network(AP) ? 

BR.

convertl, 2021-06-21:
Hi @ahhfzhang,
Thanks for your answer. Yes, both master and slave clients are on the same network and same access point for my tests.

The problem is that I don't know how to be sure to get boards with the good revision of the ESP32. The boards have been bought end of last year, while the ESP32 ECO V3 has been introduced at the beginning of 2020 I think. So there may still be lots of older ESP32 modules on LyraT mini boards in stock.

Do you think the MRM multicast error on the slave could be due to the PSRAM bug, even with the workaround?

ehabulhaq, 2021-11-10:
Hi I am trying to run the same example but I am also not able to see slave doing anything.
This is the log I get from the slave pcb. 

The master is definitely playing music and sending SYNC packets

entry 0x40080694
I (27) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (28) boot: compile time 21:36:16
I (28) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 16MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00200000
I (96) boot: End of partition table
I (100) boot_comm: chip revision: 1, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=512b0h (332464) map
I (215) esp_image: segment 1: paddr=000612d8 vaddr=3ffb0000 size=040c0h ( 16576) load
I (221) esp_image: segment 2: paddr=000653a0 vaddr=40080000 size=0ac78h ( 44152) load
I (236) esp_image: segment 3: paddr=00070020 vaddr=400d0020 size=e5ba4h (940964) map
I (518) esp_image: segment 4: paddr=00155bcc vaddr=4008ac78 size=0ea24h ( 59940) load
I (539) esp_image: segment 5: paddr=001645f8 vaddr=50000000 size=00010h (    16) load
I (554) boot: Loaded app from partition at offset 0x10000
I (554) boot: Disabling RNG early entropy source...
I (565) psram: This chip is ESP32-D0WD
I (565) spiram: Found 64MBit SPI RAM device
I (566) spiram: SPI RAM mode: flash 80m sram 80m
I (569) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (576) cpu_start: Pro cpu up.
I (579) cpu_start: Starting app cpu, entry point is 0x40081698
0x40081698: call_start_cpu1 at C:/Users/EHaq/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (1083) spiram: SPI SRAM memory test OK
I (1091) cpu_start: Pro cpu start user code
I (1091) cpu_start: cpu freq: 240000000
I (1091) cpu_start: Application information:
I (1094) cpu_start: Project name:     multi_room
I (1099) cpu_start: App version:      1
I (1104) cpu_start: Compile time:     Nov  9 2021 12:25:30
I (1110) cpu_start: ELF file SHA256:  1db5e180a94ec658...
I (1116) cpu_start: ESP-IDF:          v4.3-dirty
I (1121) heap_init: Initializing. RAM available for dynamic allocation:
I (1128) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1135) heap_init: At 3FFB5378 len 0002AC88 (171 KiB): DRAM
I (1141) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1147) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1154) heap_init: At 4009969C len 00006964 (26 KiB): IRAM
I (1160) spiram: Adding pool of 4085K of external SPI memory to heap allocator
I (1168) spi_flash: detected chip: gd
I (1172) spi_flash: flash io: qio
I (1177) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1192) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1232) MRM_EXAMPLE: Start Wi-Fi
I (1235) wifi:wifi driver task: 3ffc8920, prio:23, stack:6656, core=0
I (1235) system_api: Base MAC address is not set
I (1237) system_api: read default base MAC address from EFUSE
I (1257) wifi:wifi firmware version: c7d0450
I (1257) wifi:wifi certification version: v7.0
I (1257) wifi:config NVS flash: enabled
I (1257) wifi:config nano formating: disabled
I (1262) wifi:Init data frame dynamic rx buffer num: 32
I (1266) wifi:Init management frame dynamic rx buffer num: 32
[... 58 lines cut ...]
I (4423) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffd5f50
I (4431) ESP_AUDIO_TASK: It's a decoder
I (4436) ESP_AUDIO_TASK: 1.CUR IN:[IN_raw],CODEC:[DEC_auto],RESAMPLE:[48000],OUT:[OUT_iis],rate:0,ch:0,pos:0
I (4446) ESP_AUDIO_TASK: 2.Handles,IN:0x3f812130,CODEC:0x3f8123a0,FILTER:0x3f8127d8,OUT:0x3f812674
I (4456) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (4461) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (4467) AUDIO_PIPELINE: link el->rb, el:0x3f812130, tag:IN_raw, rb:0x3f812b28
I (4475) AUDIO_PIPELINE: link el->rb, el:0x3f8123a0, tag:DEC_auto, rb:0x3f815368
I (4483) AUDIO_PIPELINE: link el->rb, el:0x3f8127d8, tag:Audio_forge, rb:0x3f8173a8
I (4491) ESP_AUDIO_TASK: 3. Previous starting...
I (4496) AUDIO_ELEMENT: [IN_raw-0x3f812130] Element task created
W (4503) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (4511) AUDIO_ELEMENT: [DEC_auto-0x3f8123a0] Element task created
I (4518) AUDIO_ELEMENT: [DEC_auto] AEL_MSG_CMD_RESUME,state:1
I (4524) ESP_DECODER: Ready to do audio type check, pos:0, (line 104)
I (4525) ESP_AUDIO_TASK: No Blocking play with decoder
I (4537) ESP_AUDIO_TASK: Recv Element[IN_raw-0x3f812130] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (4548) ESP_AUDIO_TASK: Recv Element[IN_raw-0x3f812130] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (4560) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:771, MEM Total:4213223 Bytes, Inter:173575 Bytes, Dram:148747 Bytes

I (4572) ESP_AUDIO_CTRL: Exit play procedure, ret:0
W (4578) MRM_MULTICAST: creat addr 239.255.255.252 port 1900
W (4583) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (4591) MRM_CLIENT: Slave start searching...
W (32041) wifi:<ba-add>idx:1 (ifx:0, 10:0c:6b:16:13:e8), tid:7, ssn:0, winSize:64

binartech, 2022-01-10:
@ahhfzhang @HengYongChao 
Im working with MRM example LYRAT 4.3 and get same
(45024) MRM_MULTICAST: select error, sock = 1936876886

Any thing to check.
The Master reproduce perfectly but the salve throughs this select error

closed 2022-02-09 by espressif-bot