espressif/esp-adf #629: VOIP can not support multi-tenant？

Alpha-zh, 2021-05-26:
hi，I ran into a problem while using VOIP.
If the current logged-in user and the called user are not in the same user domain, using sip_invite() to initiate a call will automatically bring up the logged-in user's user domain, causing the call to fail.

For example.
1. The current logged-in user is: user45783080@c.sip.domain.cn
2. The call is initiated by the called user: 1001@xwx003.sip.domain.cn (not in the same domain as the login user)
3. After initiating the call, the call fails and the log is as follows (the to part is obviously incorrect)
Via: SIP/2.0/UDP 192.168.3.80:11818;branch=z9hG4bK--1044964880;rport
From: <sip:user45783080@c.sip.domain.cn>;tag=-2012458484
To: <sip:**1001@xwx003.sip.domain.cn@c.sip.domain.cn**>
Contact: <sip:user45783080@192.168.3.80:11818>

Has anyone encountered a similar problem or know how to solve it?

Translated with www.DeepL.com/Translator (free version)

Alpha-zh, 2021-05-26:
device log after invite：

I (465358) SIP: [1970-01-01/08:07:22]=======================>>
W (467368) SIP: read timeout
I (467368) SIP_MAIN: SIP_EVENT_BUSY
W (467368) SIP: CHANGE STATE FROM 64, TO 0, :func: sip_reconnect:312
I (468568) SIP: Conecting...
W (468568) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1604
I (468568) SIP: [1970-01-01/08:07:24]=======WRITE 0671 bytes>>
I (468568) SIP:

ahhfzhang, 2021-06-21:
Hi @Alpha-zh ,

We have not encountered this similar problem, but I think multi-tenant requires the server to support.

BR.

Alpha-zh, 2021-06-24:
Thanks for the reply.
The server already supports multi-tenant, the problem is that when I start calling esp_sip_uac_invite(sip,user), the parameter "user" will automatically splice the domain_name of the currently logged in user, for example, I pass "1001@xwx003. sip.domain.cn", but the actual one is "1001@xwx003.sip.domain.cn@c.sip.domain.cn"（“c.sip.domain.cn” is the login user domain name）, which leads to wrong recognition by the server and call failure.

it's a bug？ because if I pass in user with domain_name, It shouldn't splice in the domain_name of the currently logged in user anymore.

Alpha-zh, 2021-06-24:
另外说一句，现在已经通过服务端的方式规避了这个问题，大致思路是前端拨号用类似“device_mydomain_user”的形式，服务端通过增加dianplan策略，将user的domain替换为mydomain.遇到类似问题的同学可以试一下。

ahhfzhang, 2021-07-05:
Thanks for your sharing.

BR.

closed 2021-09-26 by jason-mao