espressif/esp-adf #632: VOIP on lyraT mini -  SIP_EVENT_AUDIO_SESSION_END comes before SIP_EVENT_AUDIO_SESSION_BEGIN

Patsjemoe, 2021-06-02:
Hi, 
When running VOIP on lyraT mini, I have a strange behaviour : 
When making a call, it starts first a SIP_EVENT_AUDIO_SESSION_END before starting a SIP_EVENT_AUDIO_SESSION_BEGIN.
My questions : 
1) is this programmed this way ? Or is this an error ? ( I hope an error, because I want the ESP to  go to deep sleep after the audio session ends).
2) has the SIP library changed versus ADF v2.1, because with that version, I did not have this behaviour (not mini, but lyraT or audiokit).

I (22753) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_END
W (22753) AUDIO_ELEMENT: OUT-[sip_dec] AEL_IO_ABORT
W (22753) AUDIO_ELEMENT: OUT-[filter] AEL_IO_ABORT
I (22773) SIP_RTP: send task stopped
W (22813) AUDIO_PIPELINE: There are no listener registered
I (22813) AUDIO_PIPELINE: audio_pipeline_unlinked
W (22813) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22823) AUDIO_ELEMENT: [sip_dec] Element has not create when AUDIO_ELEMENT_TER
MINATE
W (22833) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERM
INATE
W (22833) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22863) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (22863) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (22863) AUDIO_PIPELINE: There are no listener registered
I (22873) AUDIO_PIPELINE: audio_pipeline_unlinked
W (22873) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22883) AUDIO_ELEMENT: [filter] Element has not create when AUDIO_ELEMENT_TERM
INATE
W (22893) AUDIO_ELEMENT: [sip_enc] Element has not create when AUDIO_ELEMENT_TER
MINATE
W (22903) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINA
TE
W (22913) AUDIO_ELEMENT: [algo] Element has not create when AUDIO_ELEMENT_TERMIN
ATE
I (22923) SIP_RTP: receive task stopped
I (22923) SIP_RTP: RTP stopped
W (22923) SIP: CHANGE STATE FROM 72, TO 96, :func: _sip_uac_process_res_ok:1126
I (22923) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_BEGIN

HengYongChao, 2021-06-08:
@ahhfzhang  Please take a look at this question.

Patsjemoe, 2021-06-14:
@ahhfzhang 
Looking forward to know if you see the same, and if this is programmed this way, and if it should be this way? 
Regards
Ludo

Patsjemoe, 2021-06-18:
Hi,
If there is no time to look into this, can you please provide the source code of libesp_sip.a, so I can check myself.
Thanks in advance. 
Regards
Ludo

ahhfzhang, 2021-06-21:
Hi Ludo,

Hope you can provide the full logs? 

Before deep sleep did you call the esp_sip_destroy()?

BR.

Patsjemoe, 2021-06-21:
Hi Michael,
I will need to rerun for the full logs, can do that later today.
....but, the problem is also there, without running deep sleep, so you
should see the same when you run the basic VCIP.
Did you run and see the same ? This should be easy to do ?
Also, when I go to deep sleep, I did not run esp_sip_destroy, but does it
matter ? when coming out of deep sleep, it make a
full reset/start up, so I would think this is of no influence ? is this
correct ?
regards
Ludo

ahhfzhang, 2021-06-21:
Hi Ludo,

I won't reproduce the same problem, but I think SIP_EVENT_AUDIO_SESSION_END could be more than once in some scenes, why   
 not call the deep_sleep after SIP_EVENT_HANGUP? 
and yes, call the esp_sip_destroy before deep_sleep is better.

Patsjemoe, 2021-06-21:
Hi Michael,
It is strange you do not replicate this behaviour, as this is part of your
library that we cannot change.
I will go back to your original voip version without any change an re-run.
Which adf/idf do you use? I use adf v2. 2 and idf V3. 3.2
Regards
Ludo

Patsjemoe, 2021-06-21:
Hi Michael,
the VOIP program on github seems to have changed, and get now error : 
sip_start seems now to check first for wifi connection ...is this correct ?
which files do I need to add to adf v2.2 to overcome below eror ?
regards
Ludo
****************************************************************************************
Toolchain path: c:/users/ludok/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/ludok/esp/esp-idf-v3.3.2/requirements.txt are satisfied.
App "voip_app" version: v2.2-78-g7cf86c9-dirty
CC build/main/voip_app.o
C:/msys32/home/ludok/esp/esp-adf-v2.2/examples/advanced_examples/voip/main/voip_app.c:37:28: fatal error: audio_tone_uri.h: No such file or directory
compilation terminated.
make[1]: *** [/c/msys32/home/ludok/esp/esp-idf-v3.3.2/make/component_wrapper.mk:290: voip_app.o] Fout 1
make: *** [/c/msys32/home/ludok/esp/esp-idf-v3.3.2/make/project.mk:552: component-main-build] Fout 2

Patsjemoe, 2021-07-05:
> Hi Ludo,
> 
> I won't reproduce the same problem, but I think SIP_EVENT_AUDIO_SESSION_END could be more than once in some scenes, why
> not call the deep_sleep after SIP_EVENT_HANGUP?
> and yes, call the esp_sip_destroy before deep_sleep is better.

Hi Michael,
I also reproduce this behaviour when  running the latest VOIP SW with  ADFv2.3 , so it seems it is programmed this way, and you should be able to reproduce it too, as it uses the program as provided by Espressif, with lyrat Mini.
I would like to know if indeed it is intentionally, and if so, why is this done ?
regards
Ludo

Patsjemoe, 2021-07-10:
> Hi Michael,
> I also reproduce this behaviour when running the latest VOIP SW with ADFv2.3 , so it seems it is programmed this way, and you should be able to reproduce it too, as it uses the program [...]

@jason-mao : can someone please confirm if it is programmed on purpose (if so, why ?), or if this should be taken out ? 
If there is no time  to look into this, please provide the source code, so I can check myself.
Regards
Ludo

Patsjemoe, 2021-07-31:
Hi,
Any feedback pls?
Regards

closed 2021-08-04 by Patsjemoe