espressif/esp-adf #634: 8000Hz音频播放有杂音

SeaHi-Mo, 2021-06-05:
我的硬件平台：ESP32-Lyrat V4.3，在运行play_mp3 examples，指定播放8000Hz，16bits的音频时，会有杂音的情况出现。具体表现为：播放过程中出现"吧嗒吧嗒"的杂音。
高于8000Hz的音频，播放质量良好。希望能够快点解释或解决这个问题

HengYongChao, 2021-06-08:
Hi @SeaHi-Mo 

可否更换其他的 8k，16bit 的音频，交叉验证下呢？ 按理说 play_mp3 对于你是什么采样率关系不大， 或者你把可以复现的 mp3 音频上传到这里， 我们可以使用你的音频尝试复现下。

SeaHi-Mo, 2021-06-08:
我所使用的8K，16bit的音频是adf examples自带的音频文件，其路径是：adf/examples/get-started/play_mp3_rates/main/music-16b-2c-8000hz.mp3。在这之前使用了其他的8k音频，出现这个问题之后，才使用了example中的提供的音频，依旧会出现这个问题

HengYongChao, 2021-06-10:
@SeaHi-Mo  

用于演示的 8k Hz 的音频确实效果很不好， 你也可以使用其他的 8k Hz 采样率音频替代。

SeaHi-Mo, 2021-06-10:
其实我试过很多8KHz的音频，都有这个问题出现，所以我认为并不是音频本身的问题，而是ES8388对8kHz音频的编解码的兼容性问题

HengYongChao, 2021-06-15:
@SeaHi-Mo 

Thanks for your report, a bugfix of this issue will update to ADF soon.

jason-mao, 2021-09-26:
@SeaHi-Mo I think dc0e8c63dcef271103ee1800ce15e2e3e77f474d can be improve the qaulity.

closed 2021-09-26 by jason-mao