espressif/esp-adf #636: I2S recording bad with PSRAM enabled

JVKran, 2021-06-09:
Hi there!

I've encountered a very strange issue where the [example that writes an I2S recording into a WAV file on an SD card](https://github.com/espressif/esp-adf/tree/master/examples/recorder/pipeline_wav_sdcard) produces a bad recording when PSRAM is set to enabled and to be intialized on boot. 'Bad' is hard to define here; it sounds like parts are choppy. Furthermore, the file is only 1447kB while it is 1730kB when PSRAM is disabled (and the recording sounds crisp).

I tried several things. First of which was increasing the I2S DMA-Buffer Size and count. They're defaultly set to a buffer length of 300 and a count of 3. Increasing the buffer length to 1024, 500 or 800 even results in a worse recording... Changing to the ESP-IDF version 4.1 didn't change anything either. Furthermore, SPI-Flash is set to QIO 80MHz. Same goes for SPIRAM; set at 80MHz. The CPU is set at a fixed frequency of 240MHz.

This issue occurs when two I2S microphones (the CMM-4030D to be precise) and an SD-Card is connected through SPI. This board is implemented through a custom board definition. Just creating the example project, enabling PSRAM and making sure it initializes on boot should be enough to reproduce the issue. Just disabling PSRAM again results in a perfect recording.

I'm just helpless at this point and don't know where to look for further causes anymore. I'd really appreciate any pointers and hope to just've overseen something. 

Oh and by the way, I'm in need of the PSRAM since my application (not the example of course) requires the use of both Wi-Fi and Bluetooth.

Kind regards,

Jochem

HengYongChao, 2021-06-10:
Hi @JVKran 

I re-tested the example you mentioned. With PSRAM enabled, the recording result is still good.

Generally, whether PSRAM is enabled or not should have no effect on the recording quality of wav. 

My sdkconfig configuration is uploaded as follows.  [sdkconfig.txt](https://github.com/espressif/esp-adf/files/6628156/sdkconfig.txt)  , remove the txt suffix before use please.

ADF:  8121aa2336b0c491fe09e4b0ef1b211350ba2958
IDF:  722043f734fa556d66d57473ac266fb1d0ec5ad2 (built-in  adf)

JVKran, 2021-06-10:
Hi @HengYongChao,

Well, I thought enabling PSRAM wouldn't result in this behaviour either. Especially since that's what the documentation says.

Nevertheless, here we are. I tried to use the internal ESP-IDF of the ESP-ADF, but that didn't work out. I was kinda shocked that it's still at V3.3. I'm sure there's a very good reason for that, but switching to V3.3 would result in quite a lot of changes to be made to our code-base. Furthermore, I didn't even get it to work; error after error and complaining about an mconf executable that should be created after the existing IDF has been configured. Well, yeah. That's quite the rabbit-hole.

Using your SDK-Configuration file results in a slightly better recording. But still not up to par with a recording that's performed with PSRAM disabled. I guess I'll start playing with the buffer length and count again. Hopefully I'll be able to get a somewhat acceptable result without having to downgrade the entire codebase. 

Thanks for your help!

Kind regards,

Jochem

HengYongChao, 2021-06-10:
Hi  @JVKran 

I use ADF's built-in IDF v3.3 branch to compile examples, just because it is built-in, I generally use it by default to align the compilation environment and reproduce the problem. I don't have any hint or recommendation that you switch back to the v3.3 version. In fact, there are many mass-produced products that use it to work. There is nothing wrong with using it to compile.

Of course, for current newly developed cases, I strongly recommend using the latest release version IDF.

I have a question, why does it improve if you use my sdkconfig? My sdkconfig is actually the default configuration, only PSRAM is turned on.

In addition, the development board I use is lyrat v4.3. The software environment has been mentioned in the last comment.

JVKran, 2021-06-10:
Hi @HengYongChao, okay, good to know that downgrading the IDF wouldn't serve any cause.

Well, that's a good question. The difference is almost inaudible. The only difference is that the recorded file is a couple ten's of bytes longer. Could've just been a slightly faster initialization of 'something'.

Just a thought; could I just disable PSRAM initialization on boot and just manually initialize it later on? The ```audio_calloc``` function will then always use ```calloc``` instead of ```heap_caps_malloc(... MALLOC_CAP_SPIRAM ...)``` and I will have my PSRAM available for Bluetooth Wi-Fi coexistence right? Well, just checked. Audio recording is perfect again. Now with a filesize of 3457kB instead of 28.. something. So with PSRAM initializing on boot, approximately 600 bytes are just missing?

JVKran, 2021-06-10:
Well, turns out that wasn't exactly a thought-through idea. 

```
E (8382) MP3_DECODER: Allocate decoder buffer failed. bytes 27796, (line 297)
E (8382) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN,-1
W (8386) AUDIO_ELEMENT: [mp3] audio_element_on_cmd_error,7
I (8392) MP3_DECODER: Closed
W (8396) AUDIO_ELEMENT: IN-[volume] AEL_IO_ABORT
W (8408) AUDIO_ELEMENT: OUT-[file] AEL_IO_ABORT
```

Recording works, but playback can't allocate the mp3 decoder buffer.

JVKran, 2021-06-10:
Or, no wait, it was a good idea. I managed to get it to work! 

This approach (of course) requires enough non-external memory to be available and that wasn't the case. I had three audio pipelines initialized. Now, I don't know exactly how much memory one single pipeline takes up, but I imagine that can be quite a lot. Just having one initialized pipeline at the same time results in this workaround being a feasible one.

It's of course preferable for a solid solution to be found, but for now, I can at least continue working towards a first MVP. So to sum the workaround up;
1. PSRAM is enabled but not allowed to be initialized at boot.
2. PSRAM can be initialized at startup in main to allow other libraries to use external PSRAM.

Kind regards,

A (temporarily relieved) Jochem

closed 2022-08-09 by jason-mao