espressif/esp-adf #637: ADF with WROOM-32 and VSCode

periebm, 2021-06-11:
Hello guys,
I got a few questions.

I'm want to apply the equalizer to the IDF a2dp_sink example (it's already playing music from bluetooth using the internal DAC ). To do so I need to use ADF. My questions are:

-  If I copy the folders from "~esp\esp-adf\components" to "~esp\esp-idf\components", could I use the equalizer commands?

- Is there a way to use ADF with VScode, like the IDF plugin for VSCode?

-  Will the equalizer/ADF work with a regular ESP-WROOM32 board? Or do I need a Development Sound Board?

HengYongChao, 2021-06-16:
Hi @periebm 

> If I copy the folders from "~esp\esp-adf\components" to "~esp\esp-idf\components", could I use the equalizer commands?

-  ADF is built on top of IDF. I think you shouldn't use ADF as a component of IDF.

> Is there a way to use ADF with VScode, like the IDF plugin for VSCode?
- Yes,  many engineers already use VSCODE, the only thing you need to do is to set the ADF_PATH and IDF_PATH paths.

> Will the equalizer/ADF work with a regular ESP-WROOM32 board? Or do I need a Development Sound Board?
- I suggest you use the audio development board to start your project, which can save you a lot of time.

periebm, 2021-06-16:
Ok! Thank you!

closed 2021-06-16 by periebm