espressif/esp-adf #640: audio Downmix bug

mparker-landt, 2021-06-21:
On loading the example Downmix in ADF advanced examples the errors 
"too many arguments to function 'downmix_set_input_rb_timeout'"
appear if you build.

The default example expects only two parameters; audio_element_handle_t self and int ticks_to_wait. How do we pass the index as well?

HengYongChao, 2021-06-23:
Hi @mparker-landt 

Please update the ADF to latest version.
```
git checkout master
git pull
git submodule update --init --recursive
```

mparker-landt, 2021-06-23:
Hi @HengYongChao 

My adf is up to date, I keep it and the idf up to date through Github Desktop. I also tried the commands you suggested and it's still the same.

mparker-landt, 2021-06-23:
Ok, I managed to do it via the commands:
```
cd ~/esp/esp-adf
git pull
git submodule update --init --recursive
```

closed 2021-06-23 by mparker-landt