espressif/esp-adf #642: 用户自定义板级配置 my_board

Kevin-zhongyx, 2021-06-23:
Hi 请教一下，我用get-started里的play_mp3，增加了sdkconfig.defaults这个文件，内容是CONFIG_AUDIO_BOARD_CUSTOM=y。 但是编译时会报错找不到board.h。 似乎是编译依赖关系发生了问题，请问这个问题应该怎么解决？
我想用自己的板级配置，发现get-started里的play_mp3是类似的做法，但我参考后发生了和play_mp3这个example一样的问题。找不到我的board.h。

HengYongChao, 2021-06-23:
Hi @Kevin-zhongyx,
I compile the [play_mp3](https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3) example, it's ok no error, please have a try.
Here is my config [sdkconfig.zip](https://github.com/espressif/esp-adf/files/6702214/sdkconfig.zip)

Kevin-zhongyx, 2021-06-24:
[image]
我用了你的sdkconfig编译还是一样的错误，我的esp-adf是master最新的版本。commit id :51b6605396b10c4342f0c5389240c7c6643a819f

HengYongChao, 2021-06-24:
Hi @Kevin-zhongyx  
你使用的 IDF 是 ADF 內建的吗， 还是哪个版本？

Kevin-zhongyx, 2021-06-24:
是内建的 并没有改动。

Kevin-zhongyx, 2021-06-24:
你编译时用的ADF是master最新的版本吗？

HengYongChao, 2021-06-24:
把你的 adf submodule 更新完整
```
git checkout master
git pull
git submodule update --init --recursive
```

Kevin-zhongyx, 2021-06-24:
[image]

Kevin-zhongyx, 2021-06-24:
[image]
有好几个文件报找不到board.h

Kevin-zhongyx, 2021-06-24:
我检查依赖关系是 编译my_board时是会先编译audio_hal，而编译audio_hal则需要依赖audio_board，而audio_board选了CONFIG_AUDIO_BOARD_CUSTOM是没有编译任何一个文件的， 这样就会造成找不到board.h~~~

HengYongChao, 2021-06-24:
这个 issue 是修复过的， 我建议你重新克隆一份完整的 adf 再尝试。

Kevin-zhongyx, 2021-06-24:
我测试不用cmake即不用idf.py build编译  直接用make all可以编译通过。请问一下你编译通过用的cmake 版本是多少呢？
我有两台机器克隆的esp-adf都不能用idf.py build编译play_mp3，我觉得应该跟sdk的完整性关系不大，编译其它项目是正常的。

HengYongChao, 2021-06-29:
Hi @Kevin-zhongyx 

GNU make 是没有上述问题的。
我使用 cmake 可以复现你的问题，不过 cmake 我建议你使用 IDF release/v4.x 来编译。 
ADF built-in 是 v3.3.2 则使用 GNU make 编译。

closed 2021-09-26 by jason-mao