espressif/esp-adf #645: Audio on different channels

mparker-landt, 2021-06-28:
Hi,

Does anyone know if it is possible to play different audio on the left/right channels of one of the I2S on ESP32 ADF. 

E.g Track 1 on I2S0 channel right
Track 2 on I2S0 channel left

If it is what is the best example project to use as a template?

Thanks for any help on this matter.

HengYongChao, 2021-06-29:
Hi @mparker-landt 
ESP32 has two I2S peripheral interfaces, so your application can be done in principle.

HengYongChao, 2021-06-29:
Hi @mparker-landt 
Why not try [downmix pipeline process](https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/downmix_pipeline)? It can mix up to 8 channel wav format tracks to play.

mparker-landt, 2021-06-29:
So that's the method I'm trying, I've got the downmixer playing two tracks at the same time but they're being played on both left and right channels instead of track 1 on left and track 2 on the right channel.
I've tried processing the audio by using audio_element in the pipeline registration and having the process functions go to a function.

E.g for the left channel
`
static int audio_lc_process(audio_element_handle_t self, char* buf, int len, bool tag) 
{
    int rsize = audio_element_input(self, buf, len);
 
    if (len != rsize || (rsize % 4) != 0) 
    {
        ESP_LOGW(TAG, "unexpected rsize: %d, len: %d", rsize, len);
    }
 
    int16_t lSample, rSample;
    char* lSamplep = (char*) &lSample;
    char* rSamplep = (char*) &rSample;
    for (int i = 0; i < rsize; i += 4) 
    {
        rSamplep[0] = 0;
        rSamplep[1] = 0;
 
        lSamplep[0] = buf[i + 2];
        lSamplep[1] = buf[i + 3];
 
        float a = convertToF32(rSample);
        a *= 0.5;
        rSample = convertToInt16(a);
 
        buf[i] = 0;
        buf[i + 1] = 0;
 
        buf[i + 2] = lSamplep[0];
        buf[i + 3] = lSamplep[1];
    }

    rsize = audio_element_output(self, buf, rsize);
    return rsize;
}
`
However the downmixer is still playing this on both channels and I only want it on the left.

HengYongChao, 2021-06-29:
When outputting i2s stream, you just set to output one channel only.

mparker-landt, 2021-06-29:
How do I do that? I've tried changing 
`.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT `
in the I2S config and tried changing the function 
`downmix_set_out_ctx_info(downmixer, ESP_DOWNMIX_OUT_CTX_NORMAL); `
between all the left and right options and nothing seems to work. It always plays on both channels.

closed 2023-09-05 by jason-mao