espressif/esp-adf #646: A2DP music player pause delay

michalber, 2021-06-29:
Hi everyone!

I have a problem with a BT audio player based on esp_audio. When stopping audio stream from smartphone there is long delay before stream is fully closed on esp32 (about 5s) but the music is stopped immediately. Is there any possibility to compensate this delay or this is happening due to construction of BT stack?

Thanks for all replies

Wth-Esp, 2021-08-02:
Hi，@michalber

What do you mean by `fully close`?  Do you want to disconnect the BT  link between the phone and ESP32？
Could you please also add `unlink BT` after `stop audio stream`? May this delay is caused by detecting whether the connection is disconnected, the link supervision timeout of BT is typically 5s.

Thanks

michalber, 2021-08-02:
@Wth-Esp Thanks for response :)

By `fully close` I mean closing audio pipeline on ESP32 (signalized with AEL_STATUS_STATE_STOPPED). To be precise this is how logs from device looks like:
```
// Connect smartphone to device
E (2349377) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (2349377) A2DP_STREAM: A2DP bd address:, [a8:9c:ed:e1:29:e5]
I (2349387) A2DP_STREAM: A2DP audio stream configuration, codec type 0
I (2349387) A2DP_STREAM: Bluetooth configured, sample rate=44100
W (2349407) BT_APPL: new conn_srvc id:19, app_id:0
I (2349417) A2DP_STREAM: A2DP bd address:, [a8:9c:ed:e1:29:e5]
I (2349417) A2DP_STREAM: A2DP connection state =  CONNECTED

// Play music from smartphone
E (2359977) BT_APPL: bta_dm_pm_btm_status hci_status=26
I (2359977) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b
W (2359987) BT_APPL: new conn_srvc id:19, app_id:1
I (2359997) ESP_AUDIO_CTRL: Enter play procedure, src:104
I (2360007) ESP_AUDIO_CTRL: resample_rate:44100, channels:2,rate:44100
I (2360017) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe045c
I (2360017) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:713, MEM Total:2912448 Bytes, Inter:91124 Bytes, Dram:63716 Bytes

I (2360017) ESP_AUDIO_TASK: It's a decoder
I (2360037) ESP_AUDIO_TASK: 1.CUR IN:[IN_aadp],CODEC:[DEC_pcm],RESAMPLE:[44100],OUT:[OUT_iis],rate:44100, ch:2
I (2360047) ESP_AUDIO_TASK: 2.Handles,IN:0x3f8261b0,CODEC:0x3f8266b4,FILTER:0x3f922c1c,OUT:0x3f826b34
I (2360057) ESP_AUDIO_TASK: Set codec rate:44100, ch:2
I (2360067) AUDIO_PIPELINE: audio_pipeline_unlinked
W (2360067) AUDIO_PIPELINE: There are no listener registered
I (2360077) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (2360077) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (2360087) AUDIO_PIPELINE: link el->rb, el:0x3f8261b0, tag:IN_aadp, rb:0x3f92b6c4
I (2360097) AUDIO_PIPELINE: link el->rb, el:0x3f8266b4, tag:DEC_pcm, rb:0x3f922d90
I (2360107) AUDIO_PIPELINE: link el->rb, el:0x3f922c1c, tag:resample, rb:0x3f922db8
I (2360117) ESP_AUDIO_TASK: 3. Previous starting...
I (2360117) AUDIO_ELEMENT: [IN_aadp-0x3f8261b0] Element task created
I (2360127) AUDIO_THREAD: The DEC_pcm task allocate stack on external memory
I (2360137) AUDIO_ELEMENT: [DEC_pcm-0x3f8266b4] Element task created
I (2360137) AUDIO_ELEMENT: [DEC_pcm] AEL_MSG_CMD_RESUME,state:1
I (2360147) ESP_AUDIO_TASK: No Blocking MEDIA_CTRL_EVT_PLAY with decoder
I (2360147) ESP_AUDIO_CTRL: Exit play procedure, ret:0
I (2360157) RSP_FILTER: reset sample rate of source data : 44100, reset channel of source data : 2
I (2360167) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffe045c
I (2360177) ESP_AUDIO_TASK: Received muisc info then send MEDIA_CTRL_EVT_PLAY
I (2360187) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:UNKNOWN, 0
I (2360187) AUDIO_THREAD: The resample task allocate stack on external memory
I (2360197) AUDIO_ELEMENT: [resample-0x3f922c1c] Element task created
I (2360207) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (2360217) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (2360217) AUDIO_ELEMENT: [OUT_iis-0x3f826b34] Element task created
I (2360227) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (2360227) RSP_FILTER: sample rate of source data : 44100, channel of source data : 2, sample rate of destination data : 44100, channel of destination data : 2
I (2360237) I2S_STREAM: AUDIO_STREAM_WRITER
I (2360257) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_RUNNING
I (2360257) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:822, MEM Total:2893900 Bytes, Inter:90008 Bytes, Dram:62600 Bytes

// Pause music from smartphone

// After 5 to 6 seconds since pausing music on smartphone audio pipeline is stopped
I (2368997) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:3, Available:5, que:0x3ffe045c
I (2368997) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PAUSE
I (2369007) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_PAUSE
I (2369047) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_PAUSE
I (2369047) ESP_AUDIO_CTRL: Enter media_ctrl_stop procedure, src:104, type:0, st:PAUSED, phase:0
I (2369047) ESP_AUDIO_CTRL: media_ctrl_stop, src:104, type:0, st:PAUSED, phase:0, is_sent_play_cmd:0
I (2369057) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5, que:0x3ffe045c
I (2369067) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP, status:PAUSED
W (2369067) AUDIO_ELEMENT: OUT-[DEC_pcm] AEL_IO_ABORT
I (2369077) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP Done, dat:0
I (2369087) ESP_AUDIO_TASK: Received last pos: 1562621 bytes
I (2369087) ESP_AUDIO_TASK: ESP_AUDIO status is AEL_STATUS_STATE_STOPPED
I (2369097) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:822, MEM Total:2898352 Bytes, Inter:90008 Bytes, Dram:62600 Bytes

W (2369107) ESP_AUDIO_TASK: Destroy the old pipeline
W (2369117) ESP_AUDIO_TASK: The old pipeline destroyed
I (2369117) ESP_AUDIO_CTRL: Exit media_ctrl_stop procedure, ret:0
```

Is there any chance to reduce this delay?

HengYongChao, 2021-08-03:
@michalber  try use `audio_element_set_ringbuf_done(el);` to stop read element first.

michalber, 2021-08-03:
@HengYongChao 
I cannot do this, because I am stopping music directly from smartphone (e.g. Spotify app). Problematic delay is between pushing Pause button on application and pausing audio pipeline on ESP32.