espressif/esp-adf #647: esp-adf component bluetooth_service does not compile if nimble is the selected bluetooth stack

Yohannfra, 2021-07-05:
Hello,

I work on a project with esp-adf and ble. If I go with the bluedroid stack everything goes well but if I switch to nimble then when I compile, the component **esp-adf/components/bluetooth_service** doesn't compile.

I get this error
```
In file included from /Users/yohannassouline/esp/esp-adf/components/bluetooth_service/hfp_stream.c:32:
/Users/yohannassouline/esp/esp-adf/components/bluetooth_service/include/hfp_stream.h:36:10: fatal error: esp_hf_client_api.h: No such file or directory
 #include "esp_hf_client_api.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I can avoid this error by disabling the component (by commenting all the content in its CMakelists.txt) and then it compiles but I think this is a bug, that's why I'm reporting this.

Thanks

Wth-Esp, 2021-08-02:
Hi, @Yohannfra 

This should be a fault by us, sorry. `esp_hf_client_api.h` should not be included when NimBLE enabled.

We will fix it as soon as possible and if your project is urgent, please remove the ` #include "esp_hf_client_api.h"` when you use NimBLE.

Thanks

chadrockey, 2022-03-06:
There's another interaction with ESP ADF and NimBLE:

If you've set CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y

then you get the following error because a2dp isn't part of BLE:

> .espressif/esp-adf/components/bluetooth_service/bluetooth_service.c:42:10: fatal error: esp_a2dp_api.h: No such file or directory
>  #include "esp_a2dp_api.h"

Wth-Esp, 2022-03-07:
> There's another interaction with ESP ADF and NimBLE:
>
> If you've set CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
>
> then you get the following error because a2dp isn't part of BLE:

Yes. A2DP, SPP, HFP belong to Classic Bluetooth, but NimBLE is a BLE Only Bluetooth Host Stack.

chadrockey, 2022-03-07:
Yup, just a few compilation issues that prevent you from using ADF and NimBLE at the same time, even if you aren't using Bluetooth Classic audio features.  For instance, my application is using Nimble for BLE but separately reading a file and playing it through the audio pipeline.