espressif/esp-adf #648: [pipeline_a2dp_sink_and_hfp] Audio sounds strange when connecting to windows PC over HFP.

Aaron-Yang-Dev-T, 2021-07-06:
### **Environment**
> Development Kit: ESP32-LyraT
>   Kit version: v4.3
>   IDF version : v4.0.3
>   ADF version: master (76f1a1a150f9f23114ee8236d952803ce0b4a007)
>   Power Supply : USB
>   Operating System : Windows 10, version 2004 for professional
>   IDE: Visual Studio Code
>   Windows Bluetooth firmware version: LMP 8.256
>   Windows Bluetooth driver version: 22.20.0.3

### **Problem Description**
We run the example pipeline_a2dp_sink_and_hfp on ESP32-LyraT4.3, and the PC can connect to it. However, when using HFP (mSBC codec), the audio sounds strange, but when the board is connected to an android or iOS phone, the audio is correct.

  And I also try to run hfp_hf example, and send the 1k Hz audio from PC to ESP32. The wave of the received audio is not a perfect sine wave.

[image]

### **Steps to reproduce**
1. Run the pipeline_a2dp_sink_and_hfp example and connect to PC
2. Test speech

### **Debug Logs**
I added a debug message to this[ line](https://github.com/espressif/esp-idf/blob/e5f754b26cb2e77ba1ad0423b83536788c14cfc5/components/bt/host/bluedroid/btc/profile/std/hf_client/bta_hf_client_co.c#L411).

```
case OI_CODEC_SBC_CHECKSUM_MISMATCH:  
    APPL_TRACE_ERROR("error: %d", status);
```

 I found high error rate with checksum mismatch and logs as below:

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2    
load:0x3fff0018,len:4    
load:0x3fff001c,len:6904 
load:0x40078000,len:14052
load:0x40080400,len:4304
entry 0x400806e8
I (72) boot: Chip Revision: 3
I (72) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot: ESP-IDF v4.0.3-dirty 2nd stage bootloader
I (40) boot: compile time 20:26:51
I (40) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 8MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00124f80
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 3, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2f8ac (194732) map
I (181) esp_image: segment 1: paddr=0x0003f8d4 vaddr=0x3ffbdb60 size=0x0073c (  1852) load
I (182) esp_image: segment 2: paddr=0x00040018 vaddr=0x400d0018 size=0xb9910 (760080) map
0x400d0018: _stext at ??:?

I (459) esp_image: segment 3: paddr=0x000f9930 vaddr=0x3ffbe29c size=0x02d38 ( 11576) load
I (464) esp_image: segment 4: paddr=0x000fc670 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/10007097/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (466) esp_image: segment 5: paddr=0x000fca78 vaddr=0x40080400 size=0x16c34 ( 93236) load
I (528) boot: Loaded app from partition at offset 0x10000
I (528) boot: Disabling RNG early entropy source...
I (528) cpu_start: Pro cpu up.
[... 212 lines cut ...]
E (89184) BT_APPL: error: 2305
E (89274) BT_APPL: error: 2305
E (89344) BT_APPL: error: 2305
E (89394) BT_APPL: error: 2305
E (89404) BT_APPL: error: 2305
E (89444) BT_APPL: error: 2305
E (89464) BT_APPL: error: 2305
E (89514) BT_APPL: error: 2305
E (89554) BT_APPL: error: 2305
E (89564) BT_APPL: error: 2305
E (89594) BT_APPL: error: 2305
E (89744) BT_HF: APP HFP event: AUDIO_STATE_EVT
E (89744) BT_HF: --audio state disconnected
E (89744) BT_HF: bt_app_hf_client_audio_close
I (89754) BLUETOOTH_EXAMPLE: [ * ] Receive music info from Bluetooth, sample_rates=44100, bits=16, ch=2
E (89764) BT_HF: APP HFP event: VR_STATE_CHANGE_EVT
E (89764) BT_HF: --VR state disabled
I (89914) AUDIO_ELEMENT: [i2s_w] AEL_MSG_CMD_PAUSE
I (89914) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (89924) AUDIO_ELEMENT: [i2s_w] AEL_MSG_CMD_RESUME,state:4
I (89924) I2S_STREAM: AUDIO_STREAM_WRITER
I (102244) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

W (102244) BT_APPL: new conn_srvc id:19, app_id:1
```

Wth-Esp, 2021-08-02:
Hi, @Aaron-Yang-Dev-T 

We also test HFP with several laptops (or PC) and smartphones.
We also found that more PCs act strange, including the streaming audio part, but smartphones are most good enough.
After catching the air log and debug, we found that the issue is on the PC or laptop side, there are much more retransmission packets or CRC errors of the data in the air... ESP32 just cannot receive the right data.

Thanks

Aaron-Yang-Dev-T, 2021-08-04:
Hi @Wth-Esp 
   Thank you for your reply. It would be appreciated if you could provide your air log.

TVD-Contributor, 2022-03-02:
I too have observed this. I tried two Windows 10 laptops, and both played distorted audio over HFP. In both cases, the received audio was fine (from LyraT 4.3 microphone send back to the PC).

Using an Android smartphone was perfect audio in both directions.

We dropped the Bluedroid stack in favour of paying for BlueKitchen's BTStack, and this has perfect audio over HFP in both directions.

closed 2023-09-05 by jason-mao