espressif/esp-adf #649: Live Audio streaming from ESP32 to Mobile Phone using Bluetooth

shaurya0406, 2021-07-07:
I have an ESP32-WROOM module with INMP4001 I2S MIC from which I can get the I2S audio stream that I want to live stream on a mobile phone (Android App) via Bluetooth.
So I don't want a recording that gets saved as a .wav file and then send to Android. It has to be live.
Is there any example addressing this?
If not can someone guide me on how to implement that?
Thank you

HengYongChao, 2021-07-09:
Hi @shaurya0406 

Mic --> INMP4001 --> ESP32 --> Phone,  HF --> AG,  Try HFP ?

shaurya0406, 2021-07-10:
Thank you for the suggestion.
I will try it out and post my results soon.

shaurya0406, 2021-07-10:
Are there any docs on how to setup INMP441 MEMS based MIC in ESP-ADF for reading I2S Stream Input?

HengYongChao, 2021-08-10:
Hi @shaurya0406 

The MIC data currently used by ADF is collected from the codec chip. If you use i2s to connect to MIC directly, you may need to use i2s0's PDM mode.

closed 2022-02-08 by jason-mao