espressif/esp-adf #650: Can record aux-in and mic  input at the same time?

JohnJohnXia, 2021-07-08:
Change  audio_hal_codec_cfg.adc_input  = AUDIO_HAL_ADC_INPUT_LINE1 / AUDIO_HAL_ADC_INPUT_LINE2 can 
singly record mic or aux-in. But AUDIO_HAL_ADC_INPUT_ALL has no effect.

HengYongChao, 2021-07-09:
Hi @JohnJohnXia 

Which board do you use, or which codec? seems you need value the registers of codec.

JohnJohnXia, 2021-07-09:
> Hi @JohnJohnXia
> 
> Which board do you use, or which codec? seems you need value the registers of codec.

I used the ESP32-LyraT, the codec is es8388

jason-mao, 2022-02-07:
ES8388 can't be recording LIN1-LIN2 and RIN1-RIN2  at same time. But if you connect one mic to LIN1 and connect AUX-IN to RIN1, they can recoding at same time.  Please refer to ES8388 datasheet `Register 10 – ADC Control 2` for the register configuration.

closed 2022-02-07 by jason-mao