espressif/esp-adf #655: 用电脑连接ESP32的蓝牙，出现重启 (IDFGH-5566)

StonePig, 2021-07-20:
环境：
开发板：ESP32——LyraT
软件版本：ADF 搭配V4.1的IDF

1、打开电脑端的音乐播放器，用电脑来连接ESP32的蓝牙，ESP32可以正常播放出音乐。
2、在电脑端断开ESP32蓝牙，再连接蓝牙后，出现重启，log如下：

W (24801) ESP_AUDIO_CTRL: [media_ctrl_stop]-Already stopped, status is FINISHED, phase:0
I (25301) PLAYER: A2DP sink user cb
I (25301) PLAYER: audio stream transmission state changed event: param->media_ctrl_stat.cm = 2, param->audio_stat.state =W (25311) AUDIO_PIPELINE: There are no listener registered
 2
I (25311) PLAYER: User cb A2DP event: 1
W (25311) AUDIO_EVT: There is no space in external queue
W (25471) BT_HCI: reassemble_and_dispatch found unfinished packet for handle with start packet. Dropping old.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400963b9  PS      : 0x00060830  A0      : 0x80117b79  A1      : 0x3ffd2650
A2      : 0xffffffff  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x80096476  A9      : 0x3ffd2630
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x000002dd  A13     : 0x3f404b47
A14     : 0x000000fe  A15     : 0x00060323  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000003f  LBEG    : 0x400924ad  LEND    : 0x400924cf  LCOUNT  : 0x00000000

ELF file SHA256: a04374d39f031a88

Backtrace: 0x400963b6:0x3ffd2650 0x40117b76:0x3ffd2690 0x40116abd:0x3ffd26b0 0x4011ffcd:0x3ffd26d0 0x4014cdd6:0x3ffd26f0 0x4013d6da:0x3ffd2710 0x4001791d:0x3ffd2730 0x4013df19:0x3ffd2750 0x4013dff6:0x3ffd2790 0x400188f5:0x3ffd27c0 0x4013d90f:0x3ffd2800 0x40141209:0x3ffd2820 0x4008598c:0x3ffd2840 0x40019d11:0x3ffd2870 0x40055b4d:0x3ffd2890 0x4013c2eb:0x3ffd28b0 0x4013c909:0x3ffd28d0

Rebooting...
ets Jul 29 2019 12:21:46

HengYongChao, 2021-07-21:
Hi @StonePig,
针对于 BT 共存问题， Audio 这边优化一个专门的分支 [audio/stack_on_psram_v3.3](https://github.com/espressif/esp-idf/tree/audio/stack_on_psram_v3.3)， 你先用这个分支测试下你当前的项目，谢谢。 即使你需要使用 v4.x， 那么可以使用最新的 release/v4.2.2.

Wth-Esp, 2021-08-02:
你好，@StonePig 

感谢你提供的 backtrace 信息！另外，可以提供一下测试设备的型号么？这个问题我们自己并没有能够复现出来，但是我们可以找设备去尝试复现，然后请你使用其他的 release 分支进行测试，并且请向我们描述这些分支是否也有这个问题，来帮助我们缩小debug 范围，好么？

谢谢。

closed 2023-04-25 by jason-mao