espressif/esp-adf #657: No more in stream handle

keanuyaoo, 2021-07-22:
ESP_ERR_AUDIO_NO_INPUT_STREAM！！！
使用esp_audio 播放spiffs里面的音乐和http音乐源，已添加http_stream和spiffs_stream。
http音乐源可以正常播放，spiffs内的音乐源出现上面的错误，具体日志如下：
I (5630) Logic Audio: Start to play http resource file://spiffs/dingdong.mp3
I (5646) ESP_AUDIO_CTRL: Enter play procedure, src:0
I (5652) ESP_AUDIO_CTRL: Play procedure, URL is ok, src:0
I (5658) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffd1878
I (5665) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:770, MEM Total:119472 Bytes

I (5665) ESP_AUDIO_TASK: It's a decoder
I (5678) ESP_AUDIO_TASK: 1.CUR IN:[IN_file],CODEC:[DEC_mp3],RESAMPLE:[0],OUT:[OUT_iis],rate:0,ch:0,pos:0
W (5688) ESP_AUDIO_TASK: No more in stream handle
E (5693) Logic Audio: ESP_AUDIO_CALLBACK_FUNC, st:5,err:0x81001,src:0

HengYongChao, 2021-07-27:
Hi @keanuyaoo 

请阅读 example(https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_spiffs_mp3) 和按照 README.MD 来操作。

jason-mao, 2021-09-01:
@keanuyaoo 这个问题怎么样了？

closed 2022-02-07 by jason-mao