espressif/esp-adf #666: Automatic recording according to the sound level

cattleliu, 2021-08-16:
I in the record_raw_HTTP base to do according to the sound level, automatic recording of the program, I encountered a problem is the first time can be normal recording, when the environment sound small stop recording. In the next time the environment sound is relatively loud when it can not normally start.
  I'll post the key code and error report

#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_wifi.h"
#include "nvs_flash.h"

#include "esp_http_client.h"
#include "sdkconfig.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "board.h"
#include "http_stream.h"
#include "i2s_stream.h"
#include "wav_encoder.h"
#include "esp_peripherals.h"
#include "periph_button.h"
#include "periph_wifi.h"
#include "filter_resample.h"
#include "input_key_service.h"

#include "driver/gpio.h"
#include "driver/adc.h"
#include "esp_adc_cal.h"

static const char *TAG = "REC_RAW_HTTP";

#define DEMO_EXIT_BIT (BIT0)
#define DEFAULT_VREF    1100        //Use adc2_vref_to_gpio() to obtain a better estimate
#define NO_OF_SAMPLES   64  

static const adc_channel_t channel = ADC_CHANNEL_0;     //GPIO36
static const adc_bits_width_t width = ADC_WIDTH_BIT_12;
static const adc_atten_t atten = ADC_ATTEN_DB_6;
static const adc_unit_t unit = ADC_UNIT_1;
static esp_adc_cal_characteristics_t *adc_chars;

static audio_pipeline_handle_t pipeline;
//static EventGroupHandle_t EXIT_FLAG;
bool one_run_flag = true;
bool pipling_state = false;
esp_err_t _http_stream_event_handle(http_stream_event_msg_t *msg)
{
    esp_http_client_handle_t http = (esp_http_client_handle_t)msg->http_client;
    char len_buf[16];
    static int total_write = 0;

    if (msg->event_id == HTTP_STREAM_PRE_REQUEST) {
        // set header
        ESP_LOGI(TAG, "[ + ] HTTP client HTTP_STREAM_PRE_REQUEST, lenght=%d", msg->buffer_len);
        esp_http_client_set_header(http, "x-audio-sample-rates", "11025");
        esp_http_client_set_header(http, "x-audio-bits", "16");
        esp_http_client_set_header(http, "x-audio-channel", "1");
        total_write = 0;
        return ESP_OK;
    }

    if (msg->event_id == HTTP_STREAM_ON_REQUEST) {
        // write data
        int wlen = sprintf(len_buf, "%x\r\n", msg->buffer_len);
        if (esp_http_client_write(http, len_buf, wlen) <= 0) {
            return ESP_FAIL;
        }
        if (esp_http_client_write(http, msg->buffer, msg->buffer_len) <= 0) {
[... 314 lines cut ...]
W (33426) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (33436) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (33436) AUDIO_EVT: There is no space in external queue
Raw: 570        Voltage: 352mV
Raw: 206        Voltage: 195mV
Raw: 206        Voltage: 195mV
Raw: 256        Voltage: 217mV
I (52946) REC_RAW_HTTP: [ * ] pipeline run
W (52946) AUDIO_PIPELINE: Pipeline already started, state:2
Raw: 256        Voltage: 217mV
Raw: 265        Voltage: 221mV
I (62946) REC_RAW_HTTP: [ * ] pipeline run
W (62946) AUDIO_PIPELINE: Pipeline already started, state:2
Raw: 265        Voltage: 221mV
Raw: 259        Voltage: 218mV
I (72946) REC_RAW_HTTP: [ * ] pipeline run
W (72946) AUDIO_PIPELINE: Pipeline already started, state:2
Raw: 259        Voltage: 218mV
Raw: 259        Voltage: 218mV
I (82946) REC_RAW_HTTP: [ * ] pipeline run
W (82946) AUDIO_PIPELINE: Pipeline already started, state:2
Raw: 259        Voltage: 218mV
Raw: 249        Voltage: 214mV
Raw: 249        Voltage: 214mV
Raw: 188        Voltage: 188mV

HengYongChao, 2021-08-17:
Hi @cattleliu 

I saw you were using ADC to get this threshold. I suggest you use VAD API to get the sound threshold.

cattleliu, 2021-08-17:
> Hi @cattleliu
> 
> I saw you were using ADC to get this threshold. I suggest you use VAD API to get the sound threshold.

Thanks for your suggestion, I will try, actually the key is how to restart after stopping Pipeling, restart always fails.

HengYongChao, 2021-09-02:
start -- stop -- start -- stop
          |
          -- pause -- resume -- pause

closed 2022-02-07 by jason-mao