espressif/esp-adf #668: VOIP call button issue

Patsjemoe, 2021-08-18:
Hi, 
When pushing the play button in VOIP on lyrat mini, I usually get my call.
Sometimes, it seems to detect another button : 
W (7695) ADC_BTN: Old ID:3, New ID:5
W (10135) ADC_BTN: Old ID:5, New ID:3
I (10375) VOIP_EXAMPLE: [ * ] [Play] input key event
W (10375) SIP: CHANGE STATE FROM 2, TO 64, :func: esp_sip_uac_invite:2065
Only after the second (sometimes more) push, it makes the call.
Is this a matter of voltage level on the button  that is detected in the ADC ?
regards
Ludo

eslamsnono, 2021-08-19:
could you please tell me which esp-idf version you used to compile? i get error with ESP-IDF v4.4.

Patsjemoe, 2021-08-19:
Hi,
Use IDF v 3.2.2

Patsjemoe, 2021-09-12:
Hi,
Any feedback?
Regards
Ludo

HengYongChao, 2021-09-14:
@ahhfzhang Check this issue please.

jason-mao, 2022-02-08:
@Patsjemoe @eslamsnono  ADF  fix a adc_button bug at https://github.com/espressif/esp-adf/commit/df0be7a62181521f33d09dd18df49c91b0f8ab45, please help check is it solved your problem.

closed 2022-02-09 by espressif-bot