espressif/esp-adf #670: ESP-A1S won't play mp3 from sdcard

jvanlangen, 2021-08-23:
ESP-A1S won't play mp3 from sdcard.  The code is compiling, but the log while running give errors:

Running on windows 10, using visual studio code with de ESP-IDF/ESP-ADF extension.

any thoughts?

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Dev\ESP-ADF\pipeline_play_sdcard_muisc> C:\Users\jvanl\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\Users\jvanl\esp\esp-idf\tools\idf.py -p COM10 monitor
Executing action: monitor
Running idf_monitor in directory c:\dev\esp-adf\pipeline_play_sdcard_muisc
Executing "C:\Users\jvanl\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\Users\jvanl\esp\esp-idf\tools/idf_monitor.py -p COM10 -b 115200 --toolchain-prefix xtensa-esp32-elf- c:\dev\esp-adf\pipeline_play_sdcard_muisc\build\play_sdcard_muisc.elf -m 'C:\Users\jvanl\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe' 'C:\Users\jvanl\esp\esp-idf\tools\idf.py' '-p' 'COM10'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM10 instead...
--- idf_monitor on \\.\COM10 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6988
ho 0 tail 12 room 4
load:0x40078000,len:14292
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x40080678
I (28) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (28) boot: compile time 15:13:04
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (47) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=13a2ch ( 80428) map
I (141) esp_image: segment 1: paddr=00023a54 vaddr=3ffb0000 size=02a74h ( 10868) load
I (146) esp_image: segment 2: paddr=000264d0 vaddr=40080000 size=09b48h ( 39752) load
I (164) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=3213ch (205116) map
I (242) esp_image: segment 4: paddr=00062164 vaddr=40089b48 size=05650h ( 22096) load
I (251) esp_image: segment 5: paddr=000677bc vaddr=50000000 size=00010h (    16) load
I (260) boot: Loaded app from partition at offset 0x10000
I (260) boot: Disabling RNG early entropy source...
I (272) cpu_start: Pro cpu up.
I (273) cpu_start: Starting app cpu, entry point is 0x400813f4
0x400813f4: call_start_cpu1 at C:/Users/jvanl/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (287) cpu_start: Pro cpu start user code
I (287) cpu_start: cpu freq: 160000000
I (287) cpu_start: Application information:
I (291) cpu_start: Project name:     play_sdcard_muisc
I (297) cpu_start: App version:      1
I (302) cpu_start: Compile time:     Aug 23 2021 15:12:34
I (308) cpu_start: ELF file SHA256:  0f29c3e11e3a9789...
I (314) cpu_start: ESP-IDF:          v4.3-dirty
I (319) heap_init: Initializing. RAM available for dynamic allocation:
[... 32 lines cut ...]
E (1167) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1187) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1197) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1207) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1227) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1237) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1247) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1267) I2C_BUS: C:/Users/jvanl/esp/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1267) AUDIO_HAL: codec init failed!
E (1277) AUDIO_BOARD: C:/Users/jvanl/esp/esp-adf/components/audio_board/lyrat_v4_3/board.c:55 (audio_board_codec_init): Got NULL Pointer
E (1287) AUDIO_HAL: audio_hal handle is null
I (1287) PLAY_SDCARD_MUSIC: [3.0] Create audio pipeline for playback
I (1297) PLAY_SDCARD_MUSIC: [3.1] Create fatfs stream to read data from sdcard
I (1307) PLAY_SDCARD_MUSIC: [3.2] Create i2s stream to write data to codec chip
I (1337) PLAY_SDCARD_MUSIC: [3.3] Create mp3 decoder
I (1337) PLAY_SDCARD_MUSIC: [3.4] Register all elements to audio pipeline
I (1337) PLAY_SDCARD_MUSIC: [3.5] Link it together [sdcard]-->fatfs_stream-->music_decoder-->i2s_stream-->[codec_chip]
I (1357) PLAY_SDCARD_MUSIC: [3.6] Set up uri: /sdcard/test.mp3 
I (1357) PLAY_SDCARD_MUSIC: [ 4 ] Set up  event listener
I (1367) PLAY_SDCARD_MUSIC: [4.1] Listening event from all elements of pipeline
I (1377) PLAY_SDCARD_MUSIC: [4.2] Listening event from peripherals
I (1377) PLAY_SDCARD_MUSIC: [ 5 ] Start audio_pipeline
I (1387) PLAY_SDCARD_MUSIC: [ 6 ] Listen for all pipeline events
I (1397) PLAY_SDCARD_MUSIC: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
```

HengYongChao, 2021-08-24:
Hi @jvanlangen 

ESP-A1S is a self-made module by an agent, not an official espressif module. You need to pay attention to the compatibility of the PIN packaged by the module.

I see the reason of this question is that you SD card mount failed, please check it.

sajan-p-jose, 2021-10-01:
solved 
Add Ai_thinker board and  as101 driver to components
and select  Ai_thinker board from menuconfig
https://github.com/sajan-p-jose/esp-adf

closed 2022-02-07 by jason-mao