espressif/esp-adf #671: MP3 decoder buffer

fraschizzato, 2021-08-24:
Using pipeline_bt_source example on a esp32-pico-d4 results in `"MP3_DECODER: Allocate decoder buffer failed. bytes 27796, (line 297)"` . Tried different sdk configuration combination but best I van achieve is around 13000 free bytes. There's a way to reduce the MP3_DECODER required buffer?

fraschizzato, 2021-08-28:
Latest from git clone.
At this time the main error was solved by using a different device: LILYGO T-Watch-2020 (both more flash and psram).
Now I get the following error: `BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ##  ` as in https://github.com/espressif/esp-adf/issues/551 
Can't get the workaround work.

HengYongChao, 2021-08-30:
@fraschizzato 

I think you used a module without psram, and then encountered the problem of insufficient memory. Please use espressif's official audio development board to run ADF routines.

fraschizzato, 2021-08-30:
First error comex from a device without PSRAM but the second one (the Underflow) comes from a device with 8MB PSRAM.
Thanks

closed 2022-02-07 by jason-mao