espressif/esp-adf #676: a2dp_source_stream -static when I change to 24 bit audio

ridedirt, 2021-09-01:
When I change from 16 to 24 bits per sample, I receive static from Bluetooth speaker and can very slightly hear music in the background of static.

File: i2s.stream.h 
line: #define I2S_STREAM_CFG_DEFAULT()
code:
 .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,   
to
 .bits_per_sample = I2S_BITS_PER_SAMPLE_24BIT,

APBeck3rd, 2021-09-01:
The ESP-32 uses the SBC codec for streaming Bluetooth audio.   I believe that Espressif's SBC implementation does not support 24 bit audio.  Perhaps someone else knows whether or not this is true.

jason-mao, 2023-09-05:
I think the SBC is 16bit only, but the I2S support 24bti per sample.
I'm going to close inactive issue. Please reopen this if you have any questions or need any further assistance.

closed 2023-09-05 by jason-mao