espressif/esp-adf #679: ESP32S2 Internal DAC not working

vigyanabikshu1, 2021-09-12:
ESP32S2 Internal DAC not working

I tried https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_http_mp3
with internal ADC using Kaluga board

replaced the line i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT(); with
i2s_stream_cfg_t i2s_cfg = I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT();
but the following error occured

f/main/CMakeFiles/__idf_main.dir/play_http_mp3_example.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/play_http_mp3_example.c.obj   -c ../main/play_http_mp3_example.c
In file included from ../main/play_http_mp3_example.c:23:
../main/play_http_mp3_example.c: In function 'app_main':
/home/v-robotics/esp/esp-adf/components/audio_stream/include/i2s_stream.h:91:35: error: 'I2S_MODE_DAC_BUILT_IN' undeclared (first use in this function); did you mean 'I2S_OUT_DATA_BURST_EN'?
         .mode = I2S_MODE_MASTER | I2S_MODE_DAC_BUILT_IN | I2S_MODE_TX,          \
                                   ^~~~~~~~~~~~~~~~~~~~~
../main/play_http_mp3_example.c:79:32: note: in expansion of macro 'I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT'
     i2s_stream_cfg_t i2s_cfg = I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT();
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/v-robotics/esp/esp-adf/components/audio_stream/include/i2s_stream.h:91:35: note: each undeclared identifier is reported only once for each function it appears in
         .mode = I2S_MODE_MASTER | I2S_MODE_DAC_BUILT_IN | I2S_MODE_TX,          \
                                   ^~~~~~~~~~~~~~~~~~~~~
../main/play_http_mp3_example.c:79:32: note: in expansion of macro 'I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT'
     i2s_stream_cfg_t i2s_cfg = I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT();
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main/play_http_mp3_example.c:79:5: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
     i2s_stream_cfg_t i2s_cfg = I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT();
     ^~~~~~~~~~~~~~~~
In file included from /home/v-robotics/esp/esp-idf/components/driver/include/driver/i2s.h:16,
                 from /home/v-robotics/esp/esp-adf/components/audio_stream/include/i2s_stream.h:28,
                 from ../main/play_http_mp3_example.c:23:
/home/v-robotics/esp/esp-idf/components/hal/include/hal/i2s_types.h:96:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Then I tried with PDM
i2s_stream_cfg_t i2s_cfg = I2S_STREAM_TX_PDM_CFG_DEFAULT();

Now no errors, but while playing,  the scope shows a 44KHz clock in 17 and 1.4MHz clock in 18

HengYongChao, 2021-09-13:
Hi @vigyanabikshu1 

This is because ESP32-S2's I2S does not support DAC output, [pipeline_play_mp3_with_dac_or_pwm](https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_play_mp3_with_dac_or_pwm/README.md) document also mentions about it.

vigyanabikshu1, 2021-09-13:
Thank you very much for you valuable reply @HengYongChao
What about I2S_STREAM_TX_PDM_CFG_DEFAULT();  in ESP32s2? . Is it possible to play mp3 in ESP32s2 without any additional chip?

I tried with PDM
i2s_stream_cfg_t i2s_cfg = I2S_STREAM_TX_PDM_CFG_DEFAULT();
Now no errors, but no sound while playing and the scope shows a 44KHz clock in 17(DAC_1) and 1.4MHz clock in 18(DAC_2)

vigyanabikshu1, 2021-09-13:
Thank you @HengYongChao 
I succeed with PWM mode in 17 and 18 with the mentioned pipeline_play_mp3_with_dac_or_pwm example . I simply chosen PWM mode in the menuconfig. Now esp32s2 playbacks through pins directly without any I2S chips. https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_play_mp3_with_dac_or_pwm/README.md

vigyanabikshu1, 2021-09-13:
I am still wondering what happened to the PDM mode. Anyway PWM is enough for my purpose

HengYongChao, 2021-09-14:
> I am still wondering what happened to the PDM mode. Anyway PWM is enough for my purpose

Glad to hear you solve the problem! About ESP32 only I2S0 has PDM mode, and ESP32S2 do not has PDM mode.

vigyanabikshu1, 2021-09-14:
Thank you very much for your valuable suggestions.

HengYongChao, 2021-09-15:
Hi @vigyanabikshu1 , If the question has been solved, please close this issue,  you can reopen it at any time if necessary.

closed 2021-09-15 by vigyanabikshu1