espressif/esp-adf #680: ESP32 LyraT V4.3 开发板上运行ADF DLNA例程 根据输出Log打补丁失败

WF-J, 2021-09-13:
在运行DLNA例程时 根据log提示需要打补丁
 Not found right xTaskCreateRestrictedPinnedToCore. Please apply the $ADF_PATH/idf_patches/idf_v3.3_freertos.patch first

出现：
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:756
error: components/freertos/tasks.c: patch does not apply

ADF版本为 v2.3
IDF版本为  v4.3.1
请问这个怎样解决

HengYongChao, 2021-09-14:
Hi @WF-J 

这个 patch 其实是针对于 v3.3（audio/stack_on_psram） 的特殊分支的， 如果你需要在 v4.x 上面使用， 那么最近请选择  [release/v4.2.2 ](https://github.com/espressif/esp-idf/releases/tag/v4.2.2)这个分支是可以打上去的。 v4.3.1 目前打这个 patch 是有冲突的，请知悉，谢谢！

WF-J, 2021-09-14:
> Hi @WF-J
> 
> 这个 patch 其实是针对于 v3.3（audio/stack_on_psram） 的这个专门优化的分支的， 如果你需要在 v4.x 上面使用， 那么最近请选择 [release/v4.2.2 ](https://github.com/espressif/esp-idf/releases/tag/v4.2.2)这个分支是可以打上去的。 v4.3.1 目前打这个 patch 是有冲突的，请知悉，谢谢！

好的 谢谢

closed 2021-09-14 by WF-J

BFvX, 2021-11-15:
在release/4.2.2上尝试打补丁，出现了同样的错误信息，想请问可能是什么位置出现问题
MINGW64 /d/esp_env/esp-idf ((v4.2.2))
$ git apply idf_v3.3_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:756
error: components/freertos/tasks.c: patch does not apply