espressif/esp-adf #686: The wifi peripheral seems to be broken (AUD-2737) #564

jponko, 2021-09-23:
It's still broken. I don't see why this was closed since it hasn't been fixed.

jason-mao, 2021-09-26:
@jponko The #564 is fixed API changed on different IDF version. Would you please provide more details for your issue,such as IDF version ,ADF version and error logs.

jponko, 2021-09-26:
Here's my latest *failed results* for the adf example *pipeline_http_mp3
using fresh (Sept 25, 2021) installs of the IDF and ADF and following the
README.md for pipeline_http_mp3*.

__Start LOG__

***@***.***:~/esp/esp-adf/examples/player/pipeline_http_mp3$ idf.py
build
Executing action: all (aliases: build)
Running ninja in directory
/home/jponko/esp/esp-adf/examples/player/pipeline_http_mp3/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file
/home/jponko/esp/esp-adf/examples/player/pipeline_http_mp3/sdkconfig
Loading defaults file
/home/jponko/esp/esp-adf/examples/player/pipeline_http_mp3/sdkconfig.defaults...
-- App "play_http_mp3" version: v2.3-61-g6fa0292
-- Adding linker script
/home/jponko/esp/esp-adf/examples/player/pipeline_http_mp3/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script
/home/jponko/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- /home/jponko/esp/esp-adf/components/esp-adf-libs, Chip is esp32
-- Current board name is CONFIG_ESP_LYRAT_V4_3_BOARD
-- Components: adf_utils app_trace app_update asio audio_board audio_hal
audio_pipeline audio_sal audio_stream battery_service bluetooth_service
bootloader bootloader_support bt cbor clouds cmock coap console
coredump_upload_service cxx display_service driver dueros_service efuse
esp-adf-libs esp-sr esp-tls esp32 esp_actions esp_adc_cal esp_common
esp_dispatcher esp_eth esp_event esp_event_cast esp_gdbstub esp_hid
esp_http_client esp_http_server esp_https_ota esp_https_server
esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_peripherals
esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system
esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs
freemodbus freertos hal heap idf_test ieee802154 input_key_service jsmn
json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash
openssl openthread ota_service partition_table perfmon playlist protobuf-c
protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter
tinyusb tone_partition ulp unity usb vfs wear_levelling wifi_provisioning
wifi_service wpa_supplicant xtensa
-- Component paths: /home/jponko/esp/esp-adf/components/adf_utils
/home/jponko/esp/esp-idf/components/app_trace
/home/jponko/esp/esp-idf/components/app_update
/home/jponko/esp/esp-idf/components/asio
/home/jponko/esp/esp-adf/components/audio_board
/home/jponko/esp/esp-adf/components/audio_hal
/home/jponko/esp/esp-adf/components/audio_pipeline
/home/jponko/esp/esp-adf/components/audio_sal
/home/jponko/esp/esp-adf/components/audio_stream
/home/jponko/esp/esp-adf/components/battery_service
/home/jponko/esp/esp-adf/components/bluetooth_service
/home/jponko/esp/esp-idf/components/bootloader
/home/jponko/esp/esp-idf/components/bootloader_support
[... 383 lines cut ...]
AUDIO_ELEMENT_TERMINATE
W (3292) AUDIO_ELEMENT: [i2s] Element has not create when
AUDIO_ELEMENT_TERMINATE
W (3302) AUDIO_ELEMENT: [mp3] Element has not create when
AUDIO_ELEMENT_TERMINATE
W (3322) wifi:<ba-del>idx
W (3322) PERIPH_WIFI: Wi-Fi disconnected from SSID linksys, auto-reconnect
disabled, reconnect after 1000 ms
W (3322) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT,
event_id:3

__END__

*Here's what YOU REALLY need to do.*
1) Install Linux Mint and all the requirements for the idf framework, Use a
lyraT V4.3 board for testing.
2) Install idf according to the website instructions.
3) Install adf according to the website instructions.
4) Go through each and every example program in both the esp-idf and
esp-adf.
5) Fix the problem and if you can't fix it then  open a bug report with the
developers.
Note: The lyraT board Mic and Aux inputs are broken in that the sound level
outputs are unbalanced by 50% when fed equal input levels. Look at the
schematic and it's obvious why this is happening.

jason-mao, 2022-02-07:
> E (3152) esp-tls-mbedtls: No server verification option set in
esp_tls_cfg_t structure. Check esp_tls API reference
E (3152) esp-tls-mbedtls: Failed to set client configurations, returned
[0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)

The TLS have break changes,please  run `idf.py menuconfig` to select `Skip server certificate verification by default` on `ESP-TLS`

closed 2022-02-07 by jason-mao