espressif/esp-adf #691: downmix volume transition: volume drop and clicking

stwirth, 2021-10-05:
I'm using downmix to create an infinite loop of the same sound by playing a WAV file twice and transition between both smoothly.

I'm using a simple 440Hz tone like [this](https://www.mediacollege.com/audio/tone/files/440Hz_44100Hz_16bit_05sec.wav) for testing.

My problem is now that I can hear a volume decrease and also some clicking during the transition. Is the gain of both inputs increased and decreased linearly? I'd like to have a constant power cross-fade like this instead:
[image]
Is there a way to change the type of fade? Any idea on how I can avoid the clicking?

The full source code of my test can be found here: https://gist.github.com/stwirth/367fba0c07dd7bd3567be11dcf889c48

I'm using the ESP32-LyraT V4.3 with and up to date esp-adf from the master branch.

houhaiyan, 2021-10-08:
@stwirth hello, the gain increase/decrease is linearly.  Nonlinear is unsupported.
Maybe the clicking is because any channel stream is blocked before down-mix. You can increase ring-buffer waiting data time using `downmix_set_input_rb_timeout`. 
If the clicking is still happened, please save the stream after down-mix. Then send to me.

stwirth, 2021-10-08:
@houhaiyan thanks for the reply!

I set the timeout to
```
downmix_set_input_rb_timeout(downmixer, 250, INDEX_BASE_STREAM);
downmix_set_input_rb_timeout(downmixer, 250, INDEX_NEWCOME_STREAM);
```
but I can still hear the clicking.

I tried to set up a pipeline to write to a WAV file while mixing but the resulting file is always empty. My attempt is here: https://gist.github.com/stwirth/367fba0c07dd7bd3567be11dcf889c48#file-main_write-c

stwirth, 2021-10-12:
Maybe I'm making this too complicated. All I want is a click-free infinite loop of the same sound. Is there a way to tell the fatfs stream reader to start from the beginning of the file again when it gets to the end?

closed 2023-09-05 by jason-mao