espressif/esp-adf #695: Flash "play_mp3_control" error...

mianqi2016, 2021-10-07:
I have built esp-adf example - "play_mp3_control" completely:
x-special/nautilus-clipboard
copy
file:///home/mianqi/Documents/Projects/%E5%AE%89%E4%BF%A1%E5%8F%AFESP32-Audio-Kit/Screenshot%20from%202021-10-07%2019-52-55.png

while, when I flash it, it was prompted:
x-special/nautilus-clipboard
copy
file:///home/mianqi/Documents/Projects/%E5%AE%89%E4%BF%A1%E5%8F%AFESP32-Audio-Kit/Screenshot%20from%202021-10-07%2019-57-16.png

HengYongChao, 2021-10-11:
Hi @mianqi2016 

This error should be the codec chip initialization failed, and I2C bus failed to set the codec chip registers. 

Please check if your development board matches the audio HAL setting in menuconfig, the default board is Lyrat v4.3 .

mianqi2016, 2021-10-20:
@HengYongChao:
You were right, thanks for your advice, after I re-selected  "custom ... board", it was OK.