espressif/esp-adf #697: Cannot build Nimble with ADF

aldo4127, 2021-10-08:
Cannot build when Nimble is added along with pipeline_sdcard_mp3_control example. 

Below are the build errors:
fatal error: esp_a2dp_api.h: No such file or directory       #include "esp_a2dp_api.h"
fatal error: esp_hf_client_api.h: No such file or directory       #include "esp_hf_client_api.h"
fatal error: esp_bt_main.h: No such file or directory         #include "esp_bt_main.h"

I am using Visual Studio Code with ESP-IDF and ESP-ADF

Both examples works fine independently.

Also enabled BLE in menuconfig with the following sdk configurations:
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=y
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n
CONFIG_BTDM_CTRL_MODE_BTDM=n
CONFIG_BT_BLUEDROID_ENABLED=n
CONFIG_BT_NIMBLE_ENABLED=y

HengYongChao, 2021-10-11:
Hi @aldo4127 

This should be a problem with the header file inclusion between  bluedroid and nimble protocol stacks, please see question #647 for details.

jason-mao, 2022-08-09:
We fixed  on f44c5357f753a422539626a572642c540de8ff85

closed 2022-08-09 by jason-mao