espressif/esp-adf #716: Multi-room audio synchronization for esp-mdf

ehabulhaq, 2021-11-03:
Hi,

I am working on an audio system that connects multiple speakers using esp-mdf. The master node receives music via wifi or raw i2s and broadcasts it over multicast to all the node speakers. 

I am looking for an audio synchronization technique implemented for esp wifi mesh. I saw the multi-room example in adf but it is only limited to WiFi. Could you add a feature where we can synchronize using MRM over esp-mesh instead of wifi?

We are working on a product that has a huge demand in US and I would like to esp32-wrover for this example. Looking forward to hearing from you.

Thanks,
Ehab

HengYongChao, 2021-11-05:
@ahhfzhang Please check the customer needs.

ehabulhaq, 2021-12-13:
Hi,

Any updates on this issue?

I cant run the multi-room example as a client. When I click in the server button to start playing. it plays but no client plays music.

ehabulhaq, 2021-12-14:
Below is the snippet on one of the slave speakers 

it just recieves the wifi message and then nothing happens. The mrm callback is never called on the slave.

I (5069) ESP_AUDIO_CTRL: Exit play procedure, ret:0
W (5076) MRM_MULTICAST: creat addr 239.255.255.252 port 9000
W (5080) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1
I (5090) MRM_CLIENT: Slave start searching...
W (29871) wifi:<ba-add>idx:1 (ifx:0, 10:0c:6b:16:13:e8), tid:7, ssn:0, winSize:64

closed 2023-09-05 by jason-mao

ehabulhaq, 2023-09-05:
why did you close without replying to my question?

jason-mao, 2023-09-06:
Hi @ehabulhaq, sorry for my carelessness.  Is it still a problem in your project? If yes, please reopen it, if you have any other problem please create a new one.