espressif/esp-adf #717: ESP-ADF references old version of ESP-IDF

sysprogs, 2021-11-03:
As of version 2.3, the **esp-adf** references ESP-IDF 3.3.2 as a git submodule. However, installing both ESP-ADF and ESP-IDF via the online installer gets versions 2.3 and 4.3.1 respectively.

This means that simply checking out esp-adf 2.3 from Github yields a different esp-idf version than is installed via the online installer.

It would be great if each **esp-adf** release referenced the **esp-idf** version that was tested with it and is recommended to be used together.

HengYongChao, 2021-11-05:
Hi @sysprogs 

Thanks for your report! 

We have already noticed the IDF version of ADF submodule. In fact, v3.3 is still in the support period.

And we will update the ESP-IDF 3.3.2 to release v4.4 soon.

You can directly assign IDF_PATH to other versions of IDF to compile your project, such as v4.2, v4.3, even v3.3.5.

sysprogs, 2021-11-05:
Thanks for getting back to us.
We are aware of the **IDF_PATH** variable, however it requires knowing the version of ESP-IDF that is compatible with each ESP-ADF release. E.g. the ESP-IDF 3.3 referenced by ESP-ADF doesn't work with all sample projects (e.g. the **record_while_play** sample won't build out-of-the-box according to one of our users). Perhaps there is a documentation page showing which ESP-IDF version is recommended with each ESP-ADF release that we missed?

HengYongChao, 2021-11-05:
Hi @sysprogs 

ADF does have [a document mentioning](https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#about-esp-adf) the supported IDF version, and the latest MR(supplementary supported IDF version) will be merged into the master branch soon.

krzychb, 2021-11-05:
I think we need a table like https://github.com/espressif/esp-iot-solution#versions

HengYongChao, 2021-11-05:
> I think we need a table like https://github.com/espressif/esp-iot-solution#versions

It looks great, good to add the table in [$ADF_PATH/README.md](https://github.com/espressif/esp-adf/blob/master/README.md)

sysprogs, 2021-11-05:
@HengYongChao, thanks we will direct our users to that page.

closed 2021-11-05 by sysprogs