espressif/esp-adf #718: pipeline_living_stream

vinnitu, 2021-11-11:
Hello.

Trying to run pipeline_living_stream.
Use esp-idf 4.3.1

```
I (3122) HTTP_LIVINGSTREAM_EXAMPLE: [ 4 ] Set up  event listener
I (3132) HTTP_LIVINGSTREAM_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (3132) HTTP_LIVINGSTREAM_EXAMPLE: [4.2] Listening event from peripherals
I (3142) HTTP_LIVINGSTREAM_EXAMPLE: [ 5 ] Start audio_pipeline
I (3152) AUDIO_THREAD: The http task allocate stack on external memory
E (3162) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first

E (3182) AUDIO_THREAD: Error creating RestrictedPinnedToCore http
E (3182) AUDIO_ELEMENT: [http] audio_thread_create failed
I (3192) AUDIO_THREAD: The aac task allocate stack on external memory
E (3202) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first

E (3212) AUDIO_THREAD: Error creating RestrictedPinnedToCore aac
E (3222) AUDIO_ELEMENT: [aac] audio_thread_create failed
I (3232) AUDIO_ELEMENT: [i2s-0x3f800f04] Element task created
I (3232) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4285191 Bytes, Inter:144551 Bytes, Dram:123887 Bytes
```

trying apply patch

```
> git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
```

What I missed?

vinnitu, 2021-11-11:
other samples give me the same error

```
E (3162) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first

```

HengYongChao, 2021-12-03:
Hi @vinnitu 

It because we deflaut chose audio task stack run on psram, that means you need apply patch the  $ADF_PATH/idf_patches/idf_v3.3_freertos.patch` to support this feature. 

And I suggest you use idf v4.2 to compile, and this version can support the above patches.

jason-mao, 2022-02-07:
@vinnitu 
We have update the patches based on IDF version, https://github.com/espressif/esp-adf/tree/master/idf_patches.

closed 2022-02-07 by jason-mao