espressif/esp-adf #719: esp-adf still not support esp32-lyrat-mini v1.2 in menuconfig

changjian-k2k, 2021-11-13:
If there any plan in menuconfig for audio hal -> audio board supporting lyrat-mini v1.2, in which I only found lyrat-mini v1.1?

HengYongChao, 2021-11-16:
Hi @changjian-k2k 

You can still choose lyrat-mini v1.1 in the menuconfig.

The difference between those development boards is that some PCB traces are optimized.

We plan to add the audio board changes history pages for customer inquiries.

closed 2022-02-07 by jason-mao

dandandan1983, 2023-11-12:
Hi, @HengYongChao . 
Audio codec device configuration doesn't has ES8311 installed into esp32-lyrat-mini v1.2 . Is it possible to use it when i choose lyrat-mini v1.1 in the menuconfig.?
[image]