espressif/esp-adf #720: Cannot compile VoIP example with IDF v4.3.1

expresspotato, 2021-11-14:
## Environment

- Development Kit: ESP32-LyraT v4.3
- Module or chip used:  ESP32-WROVER-B
- IDF version: commit 2e74914051d14ec2290fc751a8486fb51d73a31e (HEAD, tag: v4.3.1)
- Build System: idf.py
- Operating System: Linux (ubuntu)
- Using an IDE?: Eclipse
- Power Supply: USB @ 5v

## Problem Description

Can't compile the VoIP example app using idf v4.3.1

### Expected Behavior

Be able to compile the example...

### Actual Behavior

Can't compile, fails at link time

### Steps to reproduce

1. Download esp-idf v4.3
2. Download esp-adf on master, HEAD
3. Link esp-idf in esp-adf to the one downloaded
4. cd esp-adf/examples/advanced_examples/voip
5. Compile with idf.py build

## Debug Logs

```
kevin@ubuntu:~/esp-adf/examples/advanced_examples/voip$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/kevin/esp-adf/examples/advanced_examples/voip/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/kevin/esp-adf/examples/advanced_examples/voip"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig
Loading defaults file /home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig.defaults...
/home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig.defaults:2 CONFIG_FLASHMODE_QIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_QIO
/home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig.defaults:9 CONFIG_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_SIZE
/home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig.defaults:21 CONFIG_SPIRAM_SUPPORT was replaced with CONFIG_ESP32_SPIRAM_SUPPORT
/home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig.defaults:29 CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST was replaced with CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP
/home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig.defaults:45 CONFIG_UDP_RECVMBOX_SIZE was replaced with CONFIG_LWIP_UDP_RECVMBOX_SIZE
/home/kevin/esp-adf/examples/advanced_examples/voip/sdkconfig.defaults:46 CONFIG_TCPIP_TASK_STACK_SIZE was replaced with CONFIG_LWIP_TCPIP_TASK_STACK_SIZE
-- Found PythonInterp: /home/kevin/.espressif/python_env/idf4.3_py3.8_env/bin/python (found version "3.8.10") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "voip_app" version: v2.3-71-ga481b29-dirty
-- Adding linker script /home/kevin/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/kevin/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/kevin/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/kevin/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
[... 146 lines cut ...]
esptool.py v3.1-dev
Merged 1 ELF section
Generated /home/kevin/esp-adf/examples/advanced_examples/voip/build/bootloader/bootloader.bin
[496/1108] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/esp32/touch_sensor.c.obj
/home/kevin/esp-idf/components/driver/esp32/touch_sensor.c: In function 'touch_pad_config':
/home/kevin/esp-idf/components/driver/esp32/touch_sensor.c:272:18: warning: variable 'wait_tick' set but not used [-Wunused-but-set-variable]
         uint32_t wait_tick = 0;
                  ^~~~~~~~~
[1107/1108] Linking CXX executable voip_app.elf
FAILED: voip_app.elf 
: && /home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address   CMakeFiles/voip_app.elf.dir/project_elf_src_esp32.c.obj  -o voip_app.elf  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/esp-sr/libesp-sr.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/audio_board/libaudio_board.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/dueros_service/libdueros_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/playlist/libplaylist.a  esp-idf/audio_flash_tone/libaudio_flash_tone.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/kevin/esp-adf/examples/advanced_examples/voip/build/voip_app.map  -fno-rtti  -fno-lto  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/freemodbus/libfreemodbu [... cut ...]
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/michael/work/esp/esp-adf-internal/components/esp_sip/lib/esp_sip_auth.c:53: undefined reference to `MD5Update'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/michael/work/esp/esp-adf-internal/components/esp_sip/lib/esp_sip_auth.c:54: undefined reference to `MD5Final'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

## Notes

HengYongChao, 2021-11-23:
Hi @expresspotato 

I'm follow the test conditions provided, I have compied the voip example and it is without any problems. Please check if there are other changes in your code that cause link errors. If you have any questions, please continue to add information.

I will attach a record of my compilation operation.
[voip-build-v4.3.log](https://github.com/espressif/esp-adf/files/7586571/voip-build-v4.3.log)

expresspotato, 2021-12-08:
Hi @HengYongChao,

Doesn't seem to work. You can reproduce the error with:

```
cd ~
mkdir ~/voip-test
git clone -b v4.3.1 --recursive https://github.com/espressif/esp-idf.git
git clone -b v2.3 --recursive https://github.com/espressif/esp-adf.git
cd ~/voip-test/esp-idf
./install.sh
. ./export.sh
cd ~/voip-test/esp-adf
rm -rf ./esp-idf
ln -s ~/voip-test/esp-idf esp-idf
cd ~/voip-test/esp-adf/examples/advanced_examples/voip/
idf.py build
```

...

```

/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp-adf/components/esp-adf-libs/esp_sip/lib/esp32/libesp_sip-v4x.a(esp_sip_auth.c.obj):(.literal.md5_printf+0x1c): undefined reference to `MD5Init'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp-adf/components/esp-adf-libs/esp_sip/lib/esp32/libesp_sip-v4x.a(esp_sip_auth.c.obj):(.literal.md5_printf+0x20): undefined reference to `MD5Update'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp-adf/components/esp-adf-libs/esp_sip/lib/esp32/libesp_sip-v4x.a(esp_sip_auth.c.obj):(.literal.md5_printf+0x24): undefined reference to `MD5Final'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/kevin/esp-adf/components/esp-adf-libs/esp_sip/lib/esp32/libesp_sip-v4x.a(esp_sip_auth.c.obj): in function `md5_printf':
/home/michael/work/esp/esp-adf-internal/components/esp_sip/lib/esp_sip_auth.c:52: undefined reference to `MD5Init'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/michael/work/esp/esp-adf-internal/components/esp_sip/lib/esp_sip_auth.c:53: undefined reference to `MD5Update'
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/michael/work/esp/esp-adf-internal/components/esp_sip/lib/esp_sip_auth.c:54: undefined reference to `MD5Final'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

HengYongChao, 2021-12-09:
> Hi @HengYongChao,
>
> Doesn't seem to work. You can reproduce the error with:
>
> ```
> cd ~
> mkdir ~/voip-test
> git clone -b v4.3.1 --recursive https://github.com/espressif/esp-idf.git
> git clone -b v2.3 [...]

My steps are similar to yours, I am like as follows. 

> 1. full clone adf-master 
> 2. set $ADF_PATH
> 3. full clone esp-idf release/v4.3
> 4. set IDF_PATH, ./install.sh, . ./export.sh
> 5. cd $ADF_PATH//examples/advanced_examples/voip/
> 6. idf.py build

The original build log also has operation records.

G-Pereira, 2021-12-29:
I am having the same issue, using version 3.3.2 of platformio espressif32 platform works but not 3.4.0 (latest available today)

GermanSheepDog, 2022-01-08:
I am having the same issue here, using version 4.3.2 of esp-idf on a MacBook with Monterey 12.2. I tested other version of esp.idf and every time I got same errors.

closed 2022-02-09 by espressif-bot

G-Pereira, 2022-02-09:
Disabling CONFIG_WPA_MBEDTLS_CRYPTO on sdkconfig worked for me.

viniciusmbs, 2025-06-18:
Ola gostaria de reabir o post

viniciusmbs, 2025-06-18:
Hello everyone!

My board is an ESP32 Audio Kit, version 2.2 or 2.3. I’m following this project:

https://github.com/HengYongChao/esp-adf/tree/master/components

I also checked the board documentation:

https://docs.ai-thinker.com/esp32-audio-kit

https://docs.ai-thinker.com/_media/esp32-audio-kit_v2.2_sch.pdf

I followed all the steps, especially those shared by the author of this repo. I installed Ubuntu on a virtual machine and set up everything — CMake, ESP-IDF, ADF, etc. — but I'm having trouble compiling and getting the board to work correctly.

The main issue is that it won’t connect. It seems like there’s a problem with the pins or communication. I've been working on this for over six weeks.

My goal is to create a smart intercom system where Alexa alerts me when someone rings the doorbell. I’ve already installed a door sensor, and I want to receive a notification and be able to call the board via VoIP to talk to the person at the door.

I kindly ask for help compiling this project or for a clear and updated documentation with a step-by-step guide to set it up.

I've tried several similar projects, and this one seems to be the best so far.

Thanks in advance for your help, and sorry for reopening an old topic.

P.S.: I posted the most recent comment in this thread, including the board log.