espressif/esp-adf #722: RESUME timeout

DongStop, 2021-11-16:
`// Advance to the next song when previous finishes
            if (msg.source == (void *) i2s_stream_writer
                && msg.cmd == AEL_MSG_CMD_REPORT_STATUS) {
                audio_element_state_t el_state = audio_element_get_state(i2s_stream_writer);
                if (el_state == AEL_STATE_FINISHED) {
                    ESP_LOGI(TAG, "[ * ] Finished, advancing to the next song");
                    sdcard_list_next(sdcard_list_handle, 1, &url);
                    ESP_LOGW(TAG, "URL: %s", url);
                    /* In previous versions, audio_pipeline_terminal() was called here. It will close all the elememnt task and when use
                     * the pipeline next time, all the tasks should be restart again. It speed too much time when we switch to another music.
                     * So we use another method to acheive this as below.
                     */
                    audio_element_set_uri(fatfs_stream_reader, url);
                    audio_pipeline_reset_ringbuffer(pipeline);
                    audio_pipeline_reset_elements(pipeline);
                    audio_pipeline_change_state(pipeline, AEL_STATE_INIT);
                    audio_pipeline_run(pipeline);
                }
                continue;
            }`
[image]

HengYongChao, 2021-11-23:
Hi @DongStop 

可否简要说明操作？ 看 log 似乎是上一首结束了，开始播放下一首的时候出错，此错误是偶先还是必现？

DongStop, 2021-11-25:
此问题我已解决，我认为存在的问题有可能存在于贵公司的解码库，从log信息很显然说明，当i2s steam发出AEL_STATE_FINISHED信号时，fatfs stream element 未执行close()调用。经过我单独添加了一个fatfs stream 的reset函数，添加在audio_pipeline_reset_elements()前，问题得到解决。并且，我认为贵公司该例程的写法完全是有问题的，这部分代码会导致很严重的内存泄露，每次播放下一首时，都将调用audio_pipeline_run()，请仔细查看该函数，注意该函数的行为，为每个element重新动态创建线程，但是在播放下一首时，并未退出原来element的线程，即未被destroy，在切歌的时候我添加了一段代码，查看系统的memory，切一次歌，少一段内存。

DongStop, 2021-11-25:
正确的做法是调用audio_pipeline_resume()恢复原先的线程。具体的编程细节我就不多说了。

DongStop, 2021-11-25:
该问题存在于我用来测试的部分flac歌曲，每次播放完该歌曲时，就会出现该问题。例如我所提供的log信息中，《告白气球》播放完正常播放下一首歌曲，而《蒲公英的约定》播放完无法正常播放下一首，这个时候按下reset键，重新开始，当播放到《蒲公英的约定》时，该问题重新出现，这几首歌文件信息是完整的，我使用读卡器检查了SD卡的文件数据。

DongStop, 2021-11-25:
所以我推测是flac解码的问题。

DongStop, 2021-11-25:
@HengYongChao

DongStop, 2021-11-25:
还有吐槽贵公司的校园招聘。本人应聘嵌入式开发岗位，两道编程题全例通过，结果提示被打入人才库。连一次技术交流的机会都没给。本来想借此面试机会交流IDF SDK开发时遇到问题，恐怖如斯，连面试机会都没得。

HengYongChao, 2021-11-29:
Hi @DongStop  

> 此问题我已解决，我认为存在的问题有可能存在于贵公司的解码库，从log信息很显然说明，当i2s steam发出AEL_STATE_FINISHED信号时，fatfs stream element 未执行close()调用。经过我单独添加了一个fatfs stream [...]

你好，有具体的代码实现吗，我需要去复现你说的现象。 当前我使用 [pipeline_play_sdcard_music](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_play_sdcard_music) 这个例程播放 flac 格式的音乐，未复现您提及的内存泄露问题。

HengYongChao, 2022-07-15:
@DongStop  此问题在最新 ADF 版本中无法复现，如果有需求，你可以再次更新此问题。

closed 2022-08-09 by jason-mao