espressif/esp-adf #723: VBAN Protocol for ESP ADF

tttin46, 2021-11-16:
I found that VBAN protocol is better solution compare with VOIP for simple Wifi intercom system.
[image]

Any way to modify VOIP example to VBAN 2-way audio streaming?
I just found some arduino projects:
https://github.com/rkinnett/ESP32-2-Way-Audio-Relay
https://github.com/ButschFPV/ESP32-A1S_VBAN_RECEIVER

HengYongChao, 2021-11-23:
Hi @tttin46 

Thank you for the information, we will continue to pay attention to it.

tttin46, 2021-11-23:
> Hi @tttin46
> 
> Thank you for the information, we will continue to pay attention to it.

I'm finding a way to make wireless microphone over local WIFI for talk show live streaming system. All client microphones run on Audio board will send high quality voices to PC that run Voicemeeter and Director can send voice commands to clients from this PC at the same way. A system can use much many audio boards as clients and no worry about RF interference.