espressif/esp-adf #726: 触摸按键无法触发

DongStop, 2021-11-21:
当仅使用电池供电时，不连接UART串口线，和供电线时，无法触发按键。是否未考虑共地？

HengYongChao, 2021-11-23:
Hi @DongStop  

请问是 Lyrat V4..3? GND肯定是共地的，Touch pad 无法使用的时候，有无任何异常打印？ 供电的电池电量是否充足？

如问题仍存在，请继续补充您的测试信息。

DongStop, 2021-11-25:
是的，是Lyrat V4.3 Touch pad 无法使用时不能接串口线 所以无法查看异常打印信息，但是我用串口线插入开发板，用一只手捏住U串口线USB端(金属部分)，使板子和人体"共地"时，触摸按键功能将恢复正常。所以我在考虑是不是，贵公司的工程师都是利用串口线连接时来测试的板子，而忽略了将串口线和供电线拔出后再来测试板子。从理论的角度上，开发板如果整体不接地，仅仅与电池共地时，用手按触摸按键时，触摸按键电容的电荷无法完全释放到人体中，从而无法正常触发中断。

DongStop, 2021-11-25:
@HengYongChao

HengYongChao, 2021-12-03:
Hi @DongStop 

我使用Lyrat v4.3 ， 烧录例程 [pipeline_sdcard_mp3_control](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_sdcard_mp3_control) 。 使用电池供电，没有连接任何 uart 打印，测试后 Touch Pad 可以正常使用（播放，暂停，音量加，音量减）。

我倾向于你排查的电池电量是否充足， 或者有无其他硬件链接问题， 谢谢。

jason-mao, 2022-08-09:
关闭该问题，如还有问题，请重新打开或者创建新 issue。

closed 2022-08-09 by jason-mao