espressif/esp-adf #728: VOIP - SIP: Error register device

indotronic, 2021-11-24:
SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.1.102:16244;branch=z9hG4bK--256493756;received=192.168.1.102;rport=16244
From: <sip:500@192.168.1.100:5160>;tag=-1256889527
To: <sip:500@192.168.1.100:5160>;tag=as4019de76
Call-ID: B56BD5D9249528FD92A71AECDB46BD8827D0F78D7A4A
CSeq: 9 REGISTER
Server: FPBX-15.0.16.75(16.13.0)
Allow: INVITE, ACK, CANCEL, OPTIONS, BYE, REFER, SUBSCRIBE, NOTIFY, INFO, PUBLISH, MESSAGE
Supported: replaces, timer
WWW-Authenticate: Digest algorithm=MD5, realm="asterisk", nonce="3fbcd0ac"
Content-Length: 0

I (347937) SIP: [1970-01-01/00:05:23]<<======================
I (347947) SIP: Required authentication
I (347947) SIP: [1970-01-01/00:05:23]=======WRITE 0788 bytes>>
I (347957) SIP: 

REGISTER sip:500@192.168.1.100:5160 SIP/2.0
Via: SIP/2.0/UDP 192.168.1.102:16244;branch=z9hG4bK--829994193;rport
From: <sip:500@192.168.1.100:5160>;tag=1716170607
To: <sip:500@192.168.1.100:5160>
Contact: <sip:500@192.168.1.102:16244>
Max-Forwards: 70
Call-ID: B56BD5D9249528FD92A71AECDB46BD8827D0F78D7A4A
CSeq: 10 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml
Authorization: Digest username="500", realm="asterisk", nonce="3fbcd0ac", uri="sip:192.168.1.100:5160", response="28d3b66e8cbaaa4eec68fa97e2734b39", algorithm=MD5, nc=00000001, cnonce="539f814f0d81154b"

I (348027) SIP: [1970-01-01/00:05:23]=======================>>
I (348047) SIP: [1970-01-01/00:05:23]<<=====READ 0471 bytes==
I (348047) SIP: 

SIP/2.0 403 Forbidden
Via: SIP/2.0/UDP 192.168.1.102:16244;branch=z9hG4bK--829994193;received=192.168.1.102;rport=16244
From: <sip:500@192.168.1.100:5160>;tag=1716170607
To: <sip:500@192.168.1.100:5160>;tag=as4019de76
Call-ID: B56BD5D9249528FD92A71AECDB46BD8827D0F78D7A4A
CSeq: 10 REGISTER
Server: FPBX-15.0.16.75(16.13.0)
Allow: INVITE, ACK, CANCEL, OPTIONS, BYE, REFER, SUBSCRIBE, NOTIFY, INFO, PUBLISH, MESSAGE
Supported: replaces, timer
Content-Length: 0

I (348087) SIP: [1970-01-01/00:05:23]<<======================
E (348097) SIP: Error register device
W (348097) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1682

expresspotato, 2021-12-08:
Check your registration credentials in idf.py menuconifg and in your PBX and try again.

bernardoarodrigues, 2021-12-18:
I'm getting the same error, tried with the Linphone Flexisip server and a FreePBX in Virtual Machine. Did you succeed doing it?

indotronic, 2021-12-18:
> I'm getting the same error, tried with the Linphone Free SIP Service server and a FreePBX in Virtual Machine. Did you succeed doing it?

i am solved, you just need instal esp-idf old version.

bernardoarodrigues, 2021-12-19:
Thanks for the reply, what version did you install?

ahhfzhang, 2022-01-24:
Hi , 

We found this bug on esp-idf newer version, and is related to the MD5 encryption error during register authentication.

So maybe you can set `MBEDTLS_ROM_MD5 = y` and try again, thanks. 

we will fix it later.

br.

closed 2022-02-09 by espressif-bot