espressif/esp-adf #732: VOIP: Deviations from RFCs in SIP and RTP

mihovilkolaric, 2021-12-04:
When using the advanced example VoIP, I encountered following deviations from the RFCs 3261 (SIP) and RFC4566 (SDP):

1. space between SIP-Version and CRLF
In some cases, the request-line sent from ESP32 looks like this:
```
0000   41 43 4b 20 73 69 70 3a 31 30 31 40 31 39 32 2e   ACK sip:101@192.
0010   31 36 38 2e 31 2e 32 30 36 3a 34 32 37 36 33 3b   168.1.206:42763;
0020   74 72 61 6e 73 70 6f 72 74 3d 54 43 50 20 53 49   transport=TCP SI
0030   50 2f 32 2e 30 20 0d 0a                           P/2.0 ..
```
i.e. there is a space (x20) between the request-line, and line break, which violates RFC3261. While most SIP UAs still accept it, some fail to parse the request.
Also, wireshark's dissector also does not recognize the line as SIP, and thus does not dissect its content, making it harder to analyze the SIP traffic.
The last line is expected to be:
```
0030   50 2f 32 2e 30 0d 0a                           P/2.0..
```

This happens whenever the ESP32 sends an ACK to a session establishment (i.e. after having received the 200 OK for an outgoing INVITE request). 
Note: The ACK sent by the ESP as a result to a 4xx-Error is not affected.

2. Empty line at the end of SDP
The SDP (offer and answer) is always formated as:
```
0000   76 3d 30 0d 0a 6f 3d 5a 20 30 20 33 20 49 4e 20   v=0..o=Z 0 3 IN 
0010   49 50 34 20 31 39 32 2e 31 36 38 2e 31 2e 31 36   IP4 192.168.1.16
0020   37 0d 0a 73 3d 53 49 50 20 43 61 6c 6c 0d 0a 63   7..s=SIP Call..c
0030   3d 49 4e 20 49 50 34 20 31 39 32 2e 31 36 38 2e   =IN IP4 192.168.
0040   31 2e 31 36 37 0d 0a 74 3d 30 20 30 0d 0a 6d 3d   1.167..t=0 0..m=
0050   61 75 64 69 6f 20 33 37 31 38 37 20 52 54 50 2f   audio 37187 RTP/
0060   41 56 50 20 38 20 31 30 31 0d 0a 61 3d 72 74 70   AVP 8 101..a=rtp
0070   6d 61 70 3a 38 20 50 43 4d 41 2f 38 30 30 30 0d   map:8 PCMA/8000.
0080   0a 61 3d 72 74 70 6d 61 70 3a 31 30 31 20 74 65   .a=rtpmap:101 te
0090   6c 65 70 68 6f 6e 65 2d 65 76 65 6e 74 2f 38 30   lephone-event/80
00a0   30 30 0d 0a 0d 0a                                 00....
```
i.e. ends with CRLFCRLF. According to RFC4566 it should be CRLF only (this, no empty line at the end of the SDP).
Again, most UAs accept this, but I found one very picky SDP parser, which rejects sessions with the ESP32 because of this issue.

mihovilkolaric, 2021-12-04:
btw: I could not find the sources of the implementation of esp_sip.h so I could try to fix that by myself.

ahhfzhang, 2021-12-09:
Yes, we will submit a bugfix for it, thanks for your report.

closed 2022-02-09 by espressif-bot