espressif/esp-adf #735: VOIP: Call to unknown number causes infinity REGISTER-Loop

mihovilkolaric, 2021-12-15:
With esp-adf/examples/advanced_examples/voip I observed following behavior:

When receiving a error 404 as result to an outgoing INVITE (sent by the SIP proxy when the called number is unallocated), the ESP32 logs:
```
I (38028) VOIP_EXAMPLE: SIP_EVENT_BUSY
W (38028) SIP: CHANGE STATE FROM 64, TO 0, :func: sip_reconnect:313
I (39238) SIP: Conecting...
W (39238) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1667
I (39238) SIP: [1970-01-01/00:00:19]=======WRITE 0583 bytes>>
```
(which btw. is also not fully correct, "busy" is semantically not the same as "not found").
Afterwards, ESP32 starts an infinite loop of sending REGISTER (and receiving 401) every 100-200ms, which can only be aborted by resetting the board.

The following log-snippet shows the procedure, starting from the INVITE to an unallocated number (and is stopped after a few REGISTERs).

```
W (37518) SIP: CHANGE STATE FROM 2, TO 64, :func: esp_sip_uac_invite:2065
I (37558) SIP: [1970-01-01/00:00:18]=======WRITE 0915 bytes>>
I (37558) SIP: 

INVITE sip:101@192.168.1.197:5060 SIP/2.0                                                                                                                                      
Via: SIP/2.0/UDP 192.168.1.167:17431;branch=z9hG4bK--1149141156;rport                                                                                                          
From: <sip:100@192.168.1.197:5060>;tag=-1333310726                                                                                                                             
To: <sip:101@192.168.1.197:5060>                                                                                                                                               
Contact: <sip:100@192.168.1.167:17431>                                                                                                                                         
Max-Forwards: 70                                                                                                                                                               
Call-ID: BEA28E816A1815ACCCC328F90C5EBC8A70D05FEBA005                                                                                                                          
CSeq: 3 INVITE                                                                                                                                                                 
Expires: 3600                                                                                                                                                                  
User-Agent: ESP32 SIP/2.0                                                                                                                                                      
Content-Length: 166                                                                                                                                                            
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, NOTIFY, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE                                                                                      
Content-Type: application/sdp                                                                                                                                                  
Authorization: Digest username="100", realm="192.168.1.197", nonce="Ybpbs2G6WocBmdtuHlp0btt/Uuh6JzyC", uri="sip:192.168.1.197:5060", response="74f1cdce660ca049c7bae1eb78c2e055", algorithm=md5, nc=00000001, cnonce="758c4ecae92ca4f9"                                                                                                                       

v=0                                                                                                                                                                            
o=Z 0 3 IN IP4 192.168.1.167                                                                                                                                                   
s=SIP Call                                                                                                                                                                     
c=IN IP4 192.168.1.167                                                                                                                                                         
t=0 0                                                                                                                                                                          
m=audio 31296 RTP/AVP 8 101                                                                                                                                                    
a=rtpmap:8 PCMA/8000                                                                                                                                                           
a=rtpmap:101 telephone-event/8000                                                                                                                                              

I (37638) SIP: [1970-01-01/00:00:18]=======================>>
I (37768) SIP: [1970-01-01/00:00:18]<<=====READ 0500 bytes==
I (37768) SIP: 

SIP/2.0 407 Proxy Authentication Required                                                                                                                                      
Via: SIP/2.0/UDP 192.168.1.167:17431;branch=z9hG4bK--1149141156;rport=17431;received=192.168.1.167                                                                             
From: <sip:100@192.168.1.197:5060>;tag=-1333310726                                                                                                                             
To: <sip:101@192.168.1.197:5060>;tag=df037895531a1a6ce32d36baa83329df.b88a40a9                                                                                                 
Call-ID: BEA28E816A1815ACCCC328F90C5EBC8A70D05FEBA005                                                                                                                          
CSeq: 3 INVITE                                                                                                                                                                 
Proxy-Authenticate: Digest realm="192.168.1.197", nonce="Ybpb0mG6WqYP+6+sl/XQrdoh9cDleQNL"                                                                                     
Server: kamailio (5.4.4 (x86_64/linux))                                                                                                                                        
Content-Length: 0                                                                                                                                                              

I (37818) SIP: [1970-01-01/00:00:18]<<======================
I (37828) SIP: Required ACK & Authentication
I (37828) SIP: [1970-01-01/00:00:18]=======WRITE 0432 bytes>>
I (37828) SIP: 

ACK sip:101@192.168.1.197:5060 SIP/2.0                                                                                                                                         
Via: SIP/2.0/UDP 192.168.1.167:17431;branch=z9hG4bK--1149141156;rport                                                                                                          
From: <sip:100@192.168.1.197:5060>;tag=-1333310726                                                                                                                             
To: <sip:101@192.168.1.197:5060>;tag=df037895531a1a6ce32d36baa83329df.b88a40a9                                                                                                 
Contact: <sip:100@192.168.1.167:17431>                                                                                                                                         
Max-Forwards: 70                                                                                                                                                               
Call-ID: BEA28E816A1815ACCCC328F90C5EBC8A70D05FEBA005                                                                                                                          
[... 117 lines cut ...]

I (39608) SIP: [1970-01-01/00:00:19]<<======================
I (39608) SIP: Required authentication
I (39618) SIP: [1970-01-01/00:00:19]=======WRITE 0823 bytes>>
I (39618) SIP: 

[... cut ...]
Allow-Events: presence, kpml                                                                                                                                                   
Proxy-Authorization: Digest username="100", realm="192.168.1.197", nonce="Ybpb1GG6Wqjv2lz9YejuHi0jLo/mgISM", uri="sip:192.168.1.197:5060", response="56cec5656e208011d37891d999293c3a", algorithm=md5, nc=00000001, cnonce="bf5b1719f22d41b2"                                                                                                                 

I (39698) SIP: [1970-01-01/00:00:19]=======================>>

```

kamcio2603, 2021-12-15:
+1, the same problem

mihovilkolaric, 2022-01-15:
Two further findings:
- happens for other 4xx-Response-Causes
- it does not occur when a 180 Ringing is send before the 4xx-response

Bruce213, 2022-01-18:
hi，
i met the same problem，some one solved it？

ahhfzhang, 2022-02-16:
Hi @mihovilkolaric 

Thanks for your report!

We will fix it ASAP.

br.

closed 2022-03-01 by espressif-bot