espressif/esp-adf #736: OPUS decoder random crash

huybk213, 2021-12-17:
Dear all

Sometimes, i see an randon crash in opus decoder with custom application.
Because esp-adf opus decoder is close source, I couldn't figure out. Could you please give me your suggestion?
Thanks

- SDK : esp-adf v2.3 release
- Dump

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401313bf  PS      : 0x00060630  A0      : 0x801323d4  A1      : 0x3f83e370
0x401313bf: tf_decode at /home/gitlab-runner/builds/w2jWgEzx/0/adf/esp-adf-libs-source/esp_codec/esp-opus/celt/celt_decoder.c:410

A2      : 0x00000000  A3      : 0x00000015  A4      : 0x00000000  A5      : 0x3f83e400
A6      : 0x00000060  A7      : 0x3f83e580  A8      : 0x0000199a  A9      : 0x00000001
A10     : 0xfff80000  A11     : 0xfff999a0  A12     : 0x00000000  A13     : 0x3f83e3d0
A14     : 0xffc80000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000064  LBEG    : 0x40090280  LEND    : 0x400902ae  LCOUNT  : 0x00000000
0x40090280: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x400902ae: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: f25a1f224541da06

Backtrace: 0x401313bf:0x3f83e370
0x401313bf: tf_decode at /home/gitlab-runner/builds/w2jWgEzx/0/adf/esp-adf-libs-source/esp_codec/esp-opus/celt/celt_decoder.c:410

jason-mao, 2022-01-11:
@huybk213 Thank you for your report, let  we check and reply.

houhaiyan, 2022-02-09:
@huybk213 The branch can't reproduce the crash. 
Can  you give me your music that can reproduce the crash?

closed 2022-08-09 by jason-mao