espressif/esp-adf #738: VOIP - Authentication Error

bernardoarodrigues, 2021-12-18:
Hi guys, 

I'm trying to setup voip example in a lyrat-miniV1.2, tried with the Linphone Free Sip server and got 401 Unauthorized, using the correct credentials. 
Log:

I (4676) SIP: Conecting...
W (4716) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1667
I (4716) SIP: [1970-01-01/00:00:01]=======WRITE 0643 bytes>>
I (4716) SIP: 

REGISTER sip:exampleuser@sip.linphone.org:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.86.12:19979;branch=z9hG4bK-301693656;rport
From: <sip:exampleuser@sip.linphone.org:5060>;tag=1628721025
To: <sip:exampleuser@sip.linphone.org:5060>
Contact: <sip:exampleuser@192.168.86.12:19979>
Max-Forwards: 70
Call-ID: 6E223D5BD23D4474A02AEEF2283F266C5AC910739660
CSeq: 1 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml

I (4776) SIP: [1970-01-01/00:00:01]=======================>>
I (5126) SIP: [1970-01-01/00:00:02]<<=====READ 0552 bytes==
I (5126) SIP: 

SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.86.12:19979;branch=z9hG4bK-301693656;rport=4299;received=187.73.75.49
From: <sip:exampleuser@sip.linphone.org:5060>;tag=1628721025
To: <sip:exampleuser@sip.linphone.org:5060>;tag=N7NH8jpZS1FgD
Call-ID: 6E223D5BD23D4474A02AEEF2283F266C5AC910739660
CSeq: 1 REGISTER
Server: Flexisip/2.2.0-alpha-22-g473fdaf (sofia-sip-nta/2.0)
WWW-Authenticate: Digest realm="sip.linphone.org", nonce="CK1o5QAAAABpB8bPAAAluh4OujgAAAAA", opaque="+GNywA==", algorithm=MD5, qop="auth"
Content-Length: 0

I (5176) SIP: [1970-01-01/00:00:02]<<======================
I (5186) SIP: Required authentication
I (5186) SIP: [1970-01-01/00:00:02]=======WRITE 0909 bytes>>
I (5196) SIP: 

REGISTER sip:exampleuser@sip.linphone.org:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.86.12:19979;branch=z9hG4bK--1169753162;rport
From: <sip:exampleuser@sip.linphone.org:5060>;tag=-531203804
To: <sip:exampleuser@sip.linphone.org:5060>
Contact: <sip:exampleuser@192.168.86.12:19979>
Max-Forwards: 70
Call-ID: 6E223D5BD23D4474A02AEEF2283F266C5AC910739660
CSeq: 2 REGISTER
Expires: 3600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
Allow-Events: presence, kpml
Authorization: Digest username="exampleuser", realm="sip.linphone.org", nonce="CK1o5QAAAABpB8bPAAAluh4OujgAAAAA", uri="sip:sip.linphone.org:5060", response="2277d885a3dc48441c4b6f19bdf6967a", algorithm=MD5, nc=00000001, cnonce="61388a3dd85ed02b", qop="auth"

I (5276) SIP: [1970-01-01/00:00:02]=======================>>
I (5536) SIP: [1970-01-01/00:00:02]<<=====READ 0554 bytes==
I (5536) SIP: 

SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.86.12:19979;branch=z9hG4bK--1169753162;rport=4299;received=187.73.75.49
From: <sip:exampleuser@sip.linphone.org:5060>;tag=-531203804
To: <sip:exampleuser@sip.linphone.org:5060>;tag=H4S7S8D28tZUH
Call-ID: 6E223D5BD23D4474A02AEEF2283F266C5AC910739660
CSeq: 2 REGISTER
Server: Flexisip/2.2.0-alpha-22-g473fdaf (sofia-sip-nta/2.0)
WWW-Authenticate: Digest realm="sip.linphone.org", nonce="Ca1o5QAAAACbLfjqAAC4MeY66c4AAAAA", opaque="+GNywA==", algorithm=MD5, qop="auth"
Content-Length: 0

Hasaken, 2021-12-26:
Hi guys.

I have the same issue and look forward to fixing.

Hasaken, 2021-12-28:
Resetting repository to 7105e8d commit, solves the problem. Hope this information will help you to fix the issue.

bernardoarodrigues, 2021-12-28:
Hi @Hasaken thanks for your reply will try asap, hope it'll solve the problem.

bernardoarodrigues, 2021-12-29:
Finally made it, the problem was trying to use the linphone service, I built a FreeSwitch and it worked properly.

closed 2021-12-29 by bernardoarodrigues

charon22, 2021-12-30:
Hi guys.
I also had the same problem when using the linphone service.
I want to know if this problem is the linphone service problem or the ESP problem. 
We are going to use VOIP like a regular cell phone call. 
I would like to know if this is a compatibility issue with ESP.