espressif/esp-adf #739: fail to build example "play_mp3_control" under "get_started" folder

erusyd, 2021-12-25:
esp-adf: v2.3
esp-idf: v4.3.2
mcu: esp32-s3

error info:
[981/1118] Building C object esp-idf/a...dio_hal.dir/driver/es7243/es7243.c.obj
FAILED: esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/es7243/es7243.c.obj 
/home/syd/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DADF_VER=\"\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I/home/syd/esp/esp-adf/components/audio_hal/include -I/home/syd/esp/esp-adf/components/audio_hal/driver/es8388 -I/home/syd/esp/esp-adf/components/audio_hal/driver/es8374 -I/home/syd/esp/esp-adf/components/audio_hal/driver/es8311 -I/home/syd/esp/esp-adf/components/audio_hal/driver/es7243 -I/home/syd/esp/esp-adf/components/audio_hal/driver/es7148 -I/home/syd/esp/esp-adf/components/audio_hal/driver/es7210 -I/home/syd/esp/esp-adf/components/audio_hal/driver/tas5805m -I/home/syd/esp/esp-adf/components/audio_hal/driver/zl38063 -I/home/syd/esp/esp-adf/components/audio_hal/driver/zl38063/api_lib -I/home/syd/esp/esp-adf/components/audio_hal/driver/zl38063/example_apps -I/home/syd/esp/esp-adf/components/audio_hal/driver/zl38063/firmware -I/home/syd/esp/esp-adf/components/audio_hal/driver/include -I/home/syd/esp/esp-idf/components/newlib/platform_include -I/home/syd/esp/esp-idf/components/freertos/include -I/home/syd/esp/esp-idf/components/freertos/port/xtensa/include -I/home/syd/esp/esp-idf/components/esp_hw_support/include -I/home/syd/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/syd/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/home/syd/esp/esp-idf/components/heap/include -I/home/syd/esp/esp-idf/components/log/include -I/home/syd/esp/esp-idf/components/lwip/include/apps -I/home/syd/esp/esp-idf/components/lwip/include/apps/sntp -I/home/syd/esp/esp-idf/components/lwip/lwip/src/include -I/home/syd/esp/esp-idf/components/lwip/port/esp32/include -I/home/syd/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/syd/esp/esp-idf/components/soc/include -I/home/syd/esp/esp-idf/components/soc/esp32s3/. -I/home/syd/esp/esp-idf/components/soc/esp32s3/include -I/home/syd/esp/esp-idf/components/hal/esp32s3/include -I/home/syd/esp/esp-idf/components/hal/include -I/home/syd/esp/esp-idf/components/esp_rom/include -I/home/syd/esp/esp-idf/components/esp_rom/esp32s3 -I/home/syd/esp/esp-idf/components/esp_rom/include/esp32s3 -I/home/syd/esp/esp-idf/components/esp_common/include -I/home/syd/esp/esp-idf/components/esp_system/include -I/home/syd/esp/esp-idf/components/esp32s3/include -I/home/syd/esp/esp-idf/components/driver/include -I/home/syd/esp/esp-idf/components/driver/esp32s3/include -I/home/syd/esp/esp-idf/components/esp_ringbuf/include -I/home/syd/esp/esp-idf/components/efuse/include -I/home/syd/esp/esp-idf/components/efuse/esp32s3/include -I/home/syd/esp/esp-idf/components/xtensa/include -I/home/syd/esp/esp-idf/components/xtensa/esp32s3/include -I/home/syd/esp/esp-idf/components/espcoredump/include -I/home/syd/esp/esp-idf/components/esp_timer/include -I/home/syd/esp/esp-idf/components/esp_ipc/include -I/home/syd/esp/esp-idf/components/esp_pm/include -I/home/syd/esp/esp-idf/components/vfs/include -I/home/syd/esp/esp-idf/components/esp_wifi/include -I/home/syd/esp/esp-idf/components/esp_wifi/esp32s3/include -I/home/syd/esp/esp-idf/components/esp_event/include -I/home/syd/esp/esp-idf/components/esp_netif/include -I/home/syd/esp/esp-idf/components/esp_eth/include -I/home/syd/esp/esp-idf/components/tcpip_adapter/include -I/home/syd/esp/esp-idf/components/app_trace/include -I/home/syd/esp/esp-adf/components/audio_sal/include -I/home/syd/esp/esp-adf/components/audio_board/include -I/home/syd/esp/esp-adf/components/audio_board/lyrat_v4_3 -I/home/syd/esp/esp-idf/components/mbedtls/port/include -I/home/syd/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/syd/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/syd/esp/esp-adf/components/esp_peripherals/include -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/adc_button -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/gpio_isr -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/button -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/blufi -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/IS31FL3216 -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/aw2013 -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/tca9554 -I/home/syd/esp/esp-adf/components/esp_peripherals/driver/i2c_bus -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/sdcard -I/home/syd/esp/esp-adf/components/esp_peripherals/lib/touch -I/home/syd/esp/esp-idf/components/fatfs/diskio -I/home/syd/esp/esp-idf/components/fatfs/vfs -I/home/syd/esp/esp-idf/components/fatfs/src -I/home/syd/esp/esp-idf/components/wear_levelling/include -I/home/syd/esp/esp-idf/components/spi_flash/include -I/home/syd/esp/esp-idf/components/sdmmc/include -I/home/syd/esp/esp-idf/components/console -I/home/syd/esp/esp-adf/components/audio_pipeline/include -I/home/syd/esp/esp-adf/components/esp-adf-libs/esp_audio/include -I/home/syd/esp/esp-adf/components/esp-adf-libs/esp_codec/include/codec -I/home/syd/esp/esp-adf/components/esp-adf-libs/esp_codec/include/processing -I/home/syd/esp/esp-adf/components/esp-adf-libs/media_lib_sal/include -I/home/syd/esp/esp-adf/components/esp-adf-libs/media_lib_sal/include/port -I/home/syd/esp/esp-adf/components/esp-adf-libs/recorder_engine/include -I/home/syd/esp/esp-adf/components/esp-adf-libs/audio_misc/include -I/home/syd/esp/esp-adf/components/esp-adf-libs/esp_media_protocols/include -I/home/syd/esp/esp-adf/components/esp-sr/lib/include -I/home/syd/esp/esp-adf/components/esp-sr/wake_word_engine/include -I/home/syd/esp/esp-adf/components/esp-sr/speech_command_recognition/include -I/home/syd/esp/esp-adf/components/esp-sr/acoustic_algorithm/include -I/home/syd/esp/esp-adf/components/esp-sr/esp-tts/esp_tts_chinese/include -I/home/syd/esp/esp-idf/components/nghttp/port/include -I/home/syd/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/syd/esp/esp-idf/components/tcp_transport/include -I/home/syd/esp/esp-idf/components/esp-tls -I/home/syd/esp/esp-idf/components/esp-tls/esp-tls-crypto -I/home/syd/esp/esp-idf/components/spiffs/include -I/home/syd/esp/esp-adf/components/display_service/include -I/home/syd/esp/esp-adf/components/display_service/led_indicator/include -I/home/syd/esp/esp-adf/components/display_service/led_bar/include -I/home/syd/esp/esp-adf/components/esp_dispatcher/include -I/home/syd/esp/esp-idf/components/nvs_flash/include -I/home/syd/esp/esp-idf/components/wpa_supplicant/include -I/home/syd/esp/esp-idf/components/wpa_supplicant/port/include -I/home/syd/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant -mlongcalls   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3.2\" -DESP_PLATFORM -MD -MT esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/es7243/es7243.c.obj -MF esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/es7243/es7243.c.obj.d -o esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/es7243/es7243.c.obj   -c /home/syd/esp/esp-adf/components/audio_hal/driver/es7243/es7243.c
/home/syd/esp/esp-adf/components/audio_hal/driver/es7243/es7243.c: In function 'es7243_mclk_active':
/home/syd/esp/esp-adf/components/audio_hal/driver/es7243/es7243.c:82:5: error: implicit declaration of function 'gpio_pad_select_gpio'; did you mean 'get_adc_detect_gpio'? [-Werror=implicit-function-declaration]
     gpio_pad_select_gpio(mclk_gpio);
     ^~~~~~~~~~~~~~~~~~~~
     get_adc_detect_gpio
cc1: some warnings being treated as errors
[986/1118] Building C object esp-idf/a...o_hal.dir/driver/zl38063/zl38063.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

nezen, 2021-12-29:
you may switch to idf v4.4 and give it a try

erusyd, 2021-12-30:
Succeeded. Thanks!

closed 2021-12-30 by erusyd