espressif/esp-adf #740: [example pipeline_bt_sink] 当作蓝牙音箱使用时，可以连接，无法播放，只有突突声

xupenghu, 2022-01-01:
问题描述： 我是用的硬件设`LyraT V4.3` 使用例程`pipeline_bt_sink`时，蓝牙可以连接手机，但是手机播放音乐时，只有突突声，没有音频。
1. 我怀疑是硬件问题，于是烧录了play_mp3例程，耳机是可以听到音频的，因此排除。
2. `pipeline_bt_sink`的log如下，请帮忙分析一下是什么问题
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6772
load:0x40078000,len:12072
load:0x40080400,len:6708
entry 0x40080778
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.2-107-g722043f734 2nd stage bootloader
I (39) boot: compile time 07:59:39
I (39) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00124f80
I (90) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x31254 (201300) map
I (181) esp_image: segment 1: paddr=0x0004127c vaddr=0x3ffbdb60 size=0x03284 ( 12932) load
I (187) esp_image: segment 2: paddr=0x00044508 vaddr=0x40080000 size=0x00400 (  1024) load
I (189) esp_image: segment 3: paddr=0x00044910 vaddr=0x40080400 size=0x0b700 ( 46848) load
I (217) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xa5208 (676360) map
I (454) esp_image: segment 5: paddr=0x000f5228 vaddr=0x4008bb00 size=0x0884c ( 34892) load
I (481) boot: Loaded app from partition at offset 0x10000
I (481) boot: Disabling RNG early entropy source...
I (482) cpu_start: Pro cpu up.
I (485) cpu_start: Application information:
I (490) cpu_start: Project name:     esp-idf
I (495) cpu_start: App version:      413ced2
I (500) cpu_start: Compile time:     Jan  1 2022 07:58:36
I (506) cpu_start: ELF file SHA256:  6bf5918be72255ff...
I (512) cpu_start: ESP-IDF:          v3.3.2-107-g722043f734
I (518) cpu_start: Starting app cpu, entry point is 0x40081294
I (0) cpu_start: App cpu up.
I (529) heap_init: Initializing. RAM available for dynamic allocation:
I (535) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (541) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (547) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (554) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (560) heap_init: At 3FFCFEE0 len 00010120 (64 KiB): DRAM
I (566) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (572) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (579) heap_init: At 4009434C len 0000BCB4 (47 KiB): IRAM
I (585) cpu_start: Pro cpu start user code
I (268) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (377) BLUETOOTH_EXAMPLE: [ 1 ] Create Bluetooth service
I (377) BTDM_INIT: BT controller compile version [3cd70f2]
I (387) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (507) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1127) BLUETOOTH_EXAMPLE: [ 2 ] Start codec chip
I (1137) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1157) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1157) ES8388_DRIVER: init,out:02, in:00
I (1167) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (1167) BLUETOOTH_EXAMPLE: [ 3 ] Create audio pipeline for playback
I (1167) BLUETOOTH_EXAMPLE: [3.1] Create i2s stream to write data to codec chip
I (1177) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1187) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1197) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (1207) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (1217) BLUETOOTH_EXAMPLE: [3.2] Get Bluetooth stream
I (1217) BLUETOOTH_EXAMPLE: [3.2] Register all elements to audio pipeline
I (1227) BLUETOOTH_EXAMPLE: [3.3] Link it together [Bluetooth]-->bt_stream_reader-->i2s_stream_writer-->[codec_chip]
I (1237) AUDIO_PIPELINE: link el->rb, el:0x3ffdc5d4, tag:bt, rb:0x3ffdc938
I (1247) BLUETOOTH_EXAMPLE: [ 4 ] Initialize peripherals
E (1247) gpio: gpio_install_isr_service(412): GPIO isr service already installed
I (1257) BLUETOOTH_EXAMPLE: [4.1] Initialize Touch peripheral
I (1267) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1277) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1287) BLUETOOTH_EXAMPLE: [4.2] Create Bluetooth peripheral
I (1287) BLUETOOTH_EXAMPLE: [4.2] Start all peripherals
W (1307) PERIPH_TOUCH: _touch_init
I (1307) BLUETOOTH_EXAMPLE: [ 5 ] Set up  event listener
I (1307) BLUETOOTH_EXAMPLE: [5.1] Listening event from all elements of pipeline
I (1317) BLUETOOTH_EXAMPLE: [5.2] Listening event from peripherals
I (1327) BLUETOOTH_EXAMPLE: [ 6 ] Start audio_pipeline
I (1337) AUDIO_ELEMENT: [bt-0x3ffdc5d4] Element task created
I (1347) AUDIO_ELEMENT: [i2s-0x3ffdc2b4] Element task created
I (1347) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:121568 Bytes

I (1367) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (1367) I2S_STREAM: AUDIO_STREAM_WRITER
I (1367) AUDIO_PIPELINE: Pipeline started
I (1367) BLUETOOTH_EXAMPLE: [ 7 ] Listen for all pipeline events
E (26917) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (27147) BLUETOOTH_EXAMPLE: [ * ] Receive music info from Bluetooth, sample_rates=44100, bits=16, ch=2
I (27177) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE
I (27187) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (27187) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:4
I (27197) I2S_STREAM: AUDIO_STREAM_WRITER
W (27487) BT_APPL: new conn_srvc id:19, app_id:0
I (27637) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

W (27637) BT_APPL: new conn_srvc id:19, app_id:1

```

xupenghu, 2022-01-22:
@jason-mao

HengYongChao, 2022-02-22:
Hi @xupenghu 

这个问题可以分两个方向排查下：

1. 使用 [pipeline_a2dp_sink_stream](https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream) 这个例程复测下上面是问题，是否还是存在？
2. 更换其他手机测试上面问题是够还是存在，我使用 iPhone 复测这个例程是好的。

如果有问题请继续补充，谢谢。

jason-mao, 2022-08-10:
关闭这个不活跃问题，如有需要，请重新打开或者创建新 issue。

closed 2022-08-10 by jason-mao