espressif/esp-adf #741: How to free memory after mp3 playback

vigyanabikshu1, 2022-01-03:
Free memory Before calling below Media_Play() task is 90K,
After playback complete 22K
How to free memory after playback complete ?

```
void Media_Play(void *parameter){
  // Already playing ?
  if(MEDIA_ISPLAYING){
    printf("Media Player Busy...");
    return;
  }
  
  // Get File to play
  char *currentmediafilename = (char *)parameter;
  ESP_LOGI(TAG, "Playing %s...", currentmediafilename);
  
  MEDIA_STOPPLAYBACK=0;
  currentfilesize=0;
  playbackposition=0;
  
  if(media_source == SDCARD){
    // Open renamed file for reading
    ESP_LOGI(TAG, "Reading file %s", currentmediafilename);
    currentfile = fopen(currentmediafilename, "r");
    if (currentfile == NULL) {
        ESP_LOGE(TAG, "%s Failed to open file for reading. Please check SDCARD", currentmediafilename);
        MEDIA_ISPLAYING=0;
        vTaskDelete(NULL);
        return;
    }
    
    //Get file length
    fseek(currentfile, 0, SEEK_END);
    currentfilesize = ftell(currentfile);
    fseek(currentfile, 0, SEEK_SET);	// To Begining
  }

    MEDIA_ISPLAYING=1;
    
    audio_pipeline_handle_t pipeline;
    audio_element_handle_t mp3_decoder, output_stream_writer;

    esp_log_level_set("*", ESP_LOG_WARN);
    esp_log_level_set(TAG, ESP_LOG_INFO);

    ESP_LOGI(TAG, "[ 1 ] Periph init");
    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);

    ESP_LOGI(TAG, "[ 2 ] Create audio pipeline for playback");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline = audio_pipeline_init(&pipeline_cfg);
    mem_assert(pipeline);

    ESP_LOGI(TAG, "[2.1] Create output stream to write data to codec chip");

    pwm_stream_cfg_t pwm_cfg = PWM_STREAM_CFG_DEFAULT();
    pwm_cfg.pwm_config.gpio_num_left = PWM_LEFT_OUTPUT_GPIO_NUM;
    pwm_cfg.pwm_config.gpio_num_right = PWM_RIGHT_OUTPUT_GPIO_NUM;
    output_stream_writer = pwm_stream_init(&pwm_cfg);
    ESP_LOGI(TAG, "[2.1] Create PWM output stream");

    ESP_LOGI(TAG, "[2.2] Create wav decoder to decode wav file");
    mp3_decoder_cfg_t mp3_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_decoder = mp3_decoder_init(&mp3_cfg);
    audio_element_set_read_cb(mp3_decoder, mp3_music_read_cb, NULL);

    ESP_LOGI(TAG, "[2.3] Register all elements to audio pipeline");
    audio_pipeline_register(pipeline, mp3_decoder, "mp3");
    audio_pipeline_register(pipeline, output_stream_writer, "output");

    ESP_LOGI(TAG, "[2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->output_stream-->[pa_chip]");
    const char *link_tag[2] = {"mp3", "output"};
    audio_pipeline_link(pipeline, &link_tag[0], 2);

    ESP_LOGI(TAG, "[ 3 ] Set up  event listener");
    audio_event_iface_cfg_t evt_cfg = AUDIO_EVENT_IFACE_DEFAULT_CFG();
    audio_event_iface_handle_t evt = audio_event_iface_init(&evt_cfg);

    ESP_LOGI(TAG, "[3.1] Listening event from all elements of pipeline");
    audio_pipeline_set_listener(pipeline, evt);

    ESP_LOGI(TAG, "[3.2] Listening event from peripherals");
    audio_event_iface_set_listener(esp_periph_set_get_event_iface(set), evt);

    ESP_LOGI(TAG, "[ 4 ] Start audio_pipeline");
    audio_pipeline_run(pipeline);

    ESP_LOGI(TAG, "[ 5 ] Listen for all pipeline events");
    while (1) {
        if(MEDIA_STOPPLAYBACK==1) break;
        
//        resume();
        
        //printf("\nPlaying:%d",MEDIA_ISPLAYING);
        MEDIA_ISPLAYING=1;
        audio_event_iface_msg_t msg;
        esp_err_t ret = audio_event_iface_listen(evt, &msg, portMAX_DELAY);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "[ * ] Event interface error : %d", ret);
            continue;
        }

        if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg.source == (void *) mp3_decoder
            && msg.cmd == AEL_MSG_CMD_REPORT_MUSIC_INFO) {
            audio_element_info_t music_info = {0};
            audio_element_getinfo(mp3_decoder, &music_info);
            ESP_LOGI(TAG, "[ * ] Receive music info from mp3 decoder, sample_rates=%d, bits=%d, ch=%d",
                     music_info.sample_rates, music_info.bits, music_info.channels);
            audio_element_setinfo(output_stream_writer, &music_info);
            pwm_stream_set_clk(output_stream_writer, music_info.sample_rates, music_info.bits, music_info.channels);
            continue;
        }
        /* Stop when the last pipeline element (output_stream_writer in this case) receives stop event */
        if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg.source == (void *) output_stream_writer
            && msg.cmd == AEL_MSG_CMD_REPORT_STATUS
            && (((int)msg.data == AEL_STATUS_STATE_STOPPED) || ((int)msg.data == AEL_STATUS_STATE_FINISHED))) {
            ESP_LOGI(TAG, "[ * ] Stop event received");
            break;
        }
    }
    
    MEDIA_ISPLAYING=0;
    ESP_LOGI(TAG, "[ 6 ] Stop audio_pipeline");
    audio_pipeline_stop(pipeline);
    audio_pipeline_wait_for_stop(pipeline);
    audio_pipeline_terminate(pipeline);
    audio_pipeline_unregister(pipeline, mp3_decoder);
    audio_pipeline_unregister(pipeline, output_stream_writer);

    /* Terminal the pipeline before removing the listener */
    audio_pipeline_remove_listener(pipeline);

    /* Stop all periph before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    /* Release all resources */
    audio_pipeline_deinit(pipeline);
    audio_element_deinit(mp3_decoder);
    audio_element_deinit(output_stream_writer);
    esp_periph_set_destroy(set);
    
    MEDIA_ISPLAYING=0;
    
    ESP_LOGI(TAG, "[ * ] Finished, exiting the MP3_thread");
    vTaskDelete(NULL);
}
```

NikkSaan, 2022-01-23:
maybe do `fclose` on the file? that's the only thing I see... Also, you might want to use `fatfs_stream` instead of writhing your own read callback and so on..

jason-mao, 2022-02-07:
@vigyanabikshu1 
FreeRTOS reclaims memory on idle,  recommend to call Media_Play multiple times to check memory leak.

closed 2022-08-09 by jason-mao