espressif/esp-adf #742: Can't register to linphone Sip server

charon22, 2022-01-04:
## Environment
Development Kit: ESP32-LyraT V4.3
Chip used: ESP32-WROVER-B
IDF version: v4.2.2-dirty
Build System: idf.py
Operating System: Windows 10
Using an IDE?: Power shell
Power Supply: USB / 5v

## Problem Description
Can't register to linphone Sip server with voip example
Tried with the Linphone Free Sip Service and got 401 Unauthorized, using the correct credentials.
I also checked the MICRO SIP and my phone.

Same issue : https://github.com/espressif/esp-adf/issues/738

## Question
Hi esp-adf,
I also had the same problem when using the linphone service.
I want to know if this problem is the linphone service problem or the ESP problem.
We are going to use VOIP like a regular cell phone call.
I would like to know if this is a compatibility issue with ESP.

## log:
[2021-12-14 14:54:59.512] I (10733) SIP: Conecting...
[2021-12-14 14:54:59.529] W (10753) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1667
[2021-12-14 14:54:59.529] I (10753) SIP: [1970-01-01/00:00:04]=======WRITE 0600 bytes>>
[2021-12-14 14:54:59.535] I (10753) SIP: 
[2021-12-14 14:54:59.535] 
[2021-12-14 14:54:59.535] REGISTER sip:example@sip.linphone.org:5060 SIP/2.0
[2021-12-14 14:54:59.535] Via: SIP/2.0/UDP 192.168.0.8:16025;branch=z9hG4bK-1844472995;rport
[2021-12-14 14:54:59.546] From: <sip:example@sip.linphone.org:5060>;tag=2056575315
[2021-12-14 14:54:59.552] To: <sip:example@sip.linphone.org:5060>
[2021-12-14 14:54:59.552] Contact: <sip:example@192.168.0.8:16025>
[2021-12-14 14:54:59.552] Max-Forwards: 70
[2021-12-14 14:54:59.563] Call-ID: 3C3E5C98FDE252A1035316A51195CBD427DE60D137F6
[2021-12-14 14:54:59.566] CSeq: 1 REGISTER
[2021-12-14 14:54:59.566] Expires: 3600
[2021-12-14 14:54:59.566] User-Agent: ESP32 SIP/2.0
[2021-12-14 14:54:59.566] Content-Length: 0
[2021-12-14 14:54:59.566] Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
[2021-12-14 14:54:59.579] Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
[2021-12-14 14:54:59.581] Allow-Events: presence, kpml
[2021-12-14 14:54:59.592] 
[2021-12-14 14:54:59.592] 
[2021-12-14 14:54:59.592] I (10813) SIP: [1970-01-01/00:00:04]=======================>>
[2021-12-14 14:54:59.873] I (11103) SIP: [1970-01-01/00:00:04]<<=====READ 0533 bytes==
[2021-12-14 14:54:59.876] I (11103) SIP: 
[2021-12-14 14:54:59.876] 
[2021-12-14 14:54:59.876] SIP/2.0 401 Unauthorized
[2021-12-14 14:54:59.881] Via: SIP/2.0/UDP 192.168.0.8:16025;branch=z9hG4bK-1844472995;rport=16025;received=50.68.246.146
[2021-12-14 14:54:59.890] From: <sip:example@sip.linphone.org:5060>;tag=2056575315
[2021-12-14 14:54:59.893] To: <sip:example@sip.linphone.org:5060>;tag=t735BDtgDgaDm
[2021-12-14 14:54:59.904] Call-ID: 3C3E5C98FDE252A1035316A51195CBD427DE60D137F6
[2021-12-14 14:54:59.907] CSeq: 1 REGISTER
[2021-12-14 14:54:59.907] Server: Flexisip/2.2.0-alpha-20-ga555837 (sofia-sip-nta/2.0)
[2021-12-14 14:54:59.918] WWW-Authenticate: Digest realm="sip.linphone.org", nonce="Q59j5QAAAACtrCMZAACq8bTTks4AAAAA", opaque="+GNywA==", algorithm=MD5, qop="auth"
[2021-12-14 14:54:59.928] Content-Length: 0
[2021-12-14 14:54:59.928] 
[2021-12-14 14:54:59.928] 
[2021-12-14 14:54:59.928] I (11153) SIP: [1970-01-01/00:00:04]<<======================
[2021-12-14 14:54:59.931] I (11163) SIP: Required authentication
[2021-12-14 14:54:59.944] I (11163) SIP: [1970-01-01/00:00:04]=======WRITE 0855 bytes>>
[2021-12-14 14:54:59.946] I (11173) SIP: 
[2021-12-14 14:54:59.946] 
[2021-12-14 14:54:59.946] REGISTER sip:example@sip.linphone.org:5060 SIP/2.0
[2021-12-14 14:54:59.958] Via: SIP/2.0/UDP 192.168.0.8:16025;branch=z9hG4bK--317040793;rport
[2021-12-14 14:54:59.961] From: <sip:example@sip.linphone.org:5060>;tag=-2016745904
[2021-12-14 14:54:59.961] To: <sip:example@sip.linphone.org:5060>
[2021-12-14 14:54:59.972] Contact: <sip:example@192.168.0.8:16025>
[2021-12-14 14:54:59.973] Max-Forwards: 70
[2021-12-14 14:54:59.980] Call-ID: 3C3E5C98FDE252A1035316A51195CBD427DE60D137F6
[2021-12-14 14:54:59.980] CSeq: 2 REGISTER
[2021-12-14 14:54:59.992] Expires: 3600
[2021-12-14 14:54:59.992] User-Agent: ESP32 SIP/2.0
[2021-12-14 14:54:59.992] Content-Length: 0
[2021-12-14 14:54:59.992] Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
[2021-12-14 14:54:59.996] Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
[2021-12-14 14:54:59.996] Allow-Events: presence, kpml
[2021-12-14 14:55:00.007] Authorization: Digest username="example", realm="sip.linphone.org", nonce="Q59j5QAAAACtrCMZAACq8bTTks4AAAAA", uri="sip:sip.linphone.org:5060", response="f4ab10d816a3357660e5cb158dfe6aa6", algorithm=MD5, nc=00000001, cnonce="13baa4eeb46b33f3", qop="auth"
[2021-12-14 14:55:00.025] 
[2021-12-14 14:55:00.025] 
[2021-12-14 14:55:00.025] I (11253) SIP: [1970-01-01/00:00:04]=======================>>
[2021-12-14 14:55:00.280] I (11513) SIP: [1970-01-01/00:00:04]<<=====READ 0534 bytes==
[2021-12-14 14:55:00.287] I (11513) SIP: 
[2021-12-14 14:55:00.287] 
[2021-12-14 14:55:00.287] SIP/2.0 401 Unauthorized
[2021-12-14 14:55:00.291] Via: SIP/2.0/UDP 192.168.0.8:16025;branch=z9hG4bK--317040793;rport=16025;received=50.68.246.146
[2021-12-14 14:55:00.299] From: <sip:example@sip.linphone.org:5060>;tag=-2016745904
[2021-12-14 14:55:00.309] To: <sip:example@sip.linphone.org:5060>;tag=t8cZ4eN1gDK3p
[2021-12-14 14:55:00.313] Call-ID: 3C3E5C98FDE252A1035316A51195CBD427DE60D137F6
[2021-12-14 14:55:00.324] CSeq: 2 REGISTER
[2021-12-14 14:55:00.324] Server: Flexisip/2.2.0-alpha-20-ga555837 (sofia-sip-nta/2.0)
[2021-12-14 14:55:00.326] WWW-Authenticate: Digest realm="sip.linphone.org", nonce="RJ9j5QAAAABIJ7PcAACecti9+t8AAAAA", opaque="+GNywA==", algorithm=MD5, qop="auth"
[2021-12-14 14:55:00.337] Content-Length: 0
[2021-12-14 14:55:00.340] 
[2021-12-14 14:55:00.340] 
[2021-12-14 14:55:00.340] I (11563) SIP: [1970-01-01/00:00:04]<<======================
[2021-12-14 14:55:00.351] I (11573) SIP: Required authentication
[2021-12-14 14:55:00.361] I (11573) SIP: [1970-01-01/00:00:04]=======WRITE 0853 bytes>>
[2021-12-14 14:55:00.361] I (11583) SIP: 
[2021-12-14 14:55:00.361] 
[2021-12-14 14:55:00.361] REGISTER sip:example@sip.linphone.org:5060 SIP/2.0
[2021-12-14 14:55:00.367] Via: SIP/2.0/UDP 192.168.0.8:16025;branch=z9hG4bK-891603648;rport
[2021-12-14 14:55:00.367] From: <sip:example@sip.linphone.org:5060>;tag=-377925859
[2021-12-14 14:55:00.379] To: <sip:example@sip.linphone.org:5060>
[2021-12-14 14:55:00.383] Contact: <sip:example@192.168.0.8:16025>
[2021-12-14 14:55:00.383] Max-Forwards: 70
[2021-12-14 14:55:00.383] Call-ID: 3C3E5C98FDE252A1035316A51195CBD427DE60D137F6
[2021-12-14 14:55:00.383] CSeq: 3 REGISTER
[2021-12-14 14:55:00.396] Expires: 3600
[2021-12-14 14:55:00.396] User-Agent: ESP32 SIP/2.0
[2021-12-14 14:55:00.396] Content-Length: 0
[2021-12-14 14:55:00.399] Allow: INVITE, ACK, CANCEL, BYE, UPDATE, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
[2021-12-14 14:55:00.399] Supported: replaces, norefersub, extended-refer, timer, X-cisco-serviceuri
[2021-12-14 14:55:00.412] Allow-Events: presence, kpml
[2021-12-14 14:55:00.415] Authorization: Digest username="example", realm="sip.linphone.org", nonce="RJ9j5QAAAABIJ7PcAACecti9+t8AAAAA", uri="sip:sip.linphone.org:5060", response="ae12933813e7937980c7d267a8c2a9c7", algorithm=MD5, nc=00000001, cnonce="13c9984540fd0fe1", qop="auth"
[2021-12-14 14:55:00.429] 
[2021-12-14 14:55:00.429] 
[2021-12-14 14:55:00.429] I (11653) SIP: [1970-01-01/00:00:04]=======================>>
[2021-12-14 14:55:00.795] I (12033) SIP: [1970-01-01/00:00:05]<<=====READ 0532 bytes==
[2021-12-14 14:55:00.802] I (12033) SIP: 
[2021-12-14 14:55:00.802] 
[2021-12-14 14:55:00.802] SIP/2.0 401 Unauthorized

Hasaken, 2022-01-11:
Dear ADF Support Team,

We face the same issue on the latest release. We downgrade to "ea66e72c5f82baadeb55f986176fbae4032fe1b0" commit to keep our project working. We use own SIP server and TLS connection.
We tried to use the latest revision and newly added parameters:

CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y

with no success.

Bruce213, 2022-01-18:
Hi，
Has this problem been solved? 
I had the same problem with Myvoip SIP Server，
can anyone give some help？

Hasaken, 2022-01-18:
Hi Bruce213.

The problem is still there.

Have you tried "ea66e72c5f82baadeb55f986176fbae4032fe1b0" commit?

charon22, 2022-01-18:
Hi Bruce213.
Still not resolved.

Bruce213, 2022-01-19:
hi，
I can register esp32 to my sip server now，just change idf version to v3.3.
But the registration is not very stable, after a period of time it will become offline on sip server，but sip registration  ack is OK.

ahhfzhang, 2022-01-24:
Hi charon22， 

Sorry, We don't support linphone sip server yet, you can try FreeSWITCH or FreePBX.

br.

ahhfzhang, 2022-01-24:
Hi Hasaken,

We will fix the TLS connection issue which is happening on the latest IDF release version.

br.

charon22, 2022-01-24:
Hi ahhfzhang,

We were wondering if there is any compatibility according to the VoIP server, 
so we took the Linphone server as an example. 
We ultimately want to provide users with VoIP services instead of regular phones. 
For this, we want to use the twilio Voip server(www.twilio.com) to get a VoIP number. 
we would like to know if this VoIP server is supported.

Thanks

closed 2022-02-09 by espressif-bot

charon22, 2022-02-16:
Hi Ahhfzhang,

Thank you for the reply. 
We're going to do a test. 

Thanks.

charon22, 2022-03-10:
Hi Ahhfzhang,

I made the Twilio account free and created a test account and tested it. 
No connection. 
So, after sending the ESP log to Twilio, I received the following answer as a result of inquiring. 
Please analyze this.
ESP32 LOG)
[teraterm_COM6_20220309_153824.log.TXT](https://github.com/espressif/esp-adf/files/8225838/teraterm_COM6_20220309_153824.log.TXT)

Twilio Response)
I am seeing multiple problems with the ESP32. Firstly, the ESP32 has a Request-URI which includes the user portion:
 
REGISTER sip:test@epicsafety.sip.twilio.com:5060 SIP/2.0
 
However, as per RFC 3261 section 10.2: Request-URI: The Request-URI names the domain of the location service for which the registration is meant (for example, "sip:[chicago.com](http://chicago.com/)"). The "userinfo" and "@" components of the SIP URI MUST NOT be present.
 
This means that the ESP32 should be sending REGISTER sip:epicsafety.sip.twilio.com SIP/2.0 <-- this is from a working registration.
 
I also noticed that the digest response is incorrect from the ESP32. 
This means that the ESP32 is doing something wrong when calculating the digest. 
Here is a quick summary of the calculation:
---------------------------------------------------------------------------------------------------------
HA1 = MD5(test:sip.twilio.com:Aaabbbcccdd1)
so, HA1 = 79b120bd899daf3d07c16be8e8c21d5a

HA2 = MD5(REGISTER:sip:epicsafety.sip.twilio.com)
so, HA2 = be16169f6b30b458e9534e741fd08174

Response = MD5(HA1:server_nonce:nc:cnonce:qop:HA2)
In other words, Response = MD5(79b120bd899daf3d07c16be8e8c21d5a:0WjLsbF6k7Z-ogcfJMIpsBxRoxu-AHBPPL31CQpy4MJ1lQal:00000001:9936de1b03473cf8:auth:be16169f6b30b458e9534e741fd08174)
In the above, server_nounce is the nounce that Twilio sent and cnounce is the nounce that your client generated
so, Response = fb3fd3f656959f84e7ca1e71948b0284, however, you ESP32 sent 7f2b20215d0c0a51eafeb3a9bf729e1f.
 
I don't know why your ESP32 is not calculating the correct digest response

Thanks
Sungsu