espressif/esp-adf #743: examples: pipeline_tcp_client

chitarrastonata, 2022-01-09:
Hi, i found some troubles using this example:
1) README.md say to use the tool : tcp_server_write.py. The filename in the 'tools' folder is called: tcp_server.py
2) board is connecting correctly to WiFi. Create a TCP connection to the server (tcp_server.py). Receive the MP3 information. But no music is played.
I tried to use a TCP client tool (hercules) to the python server, and connection and the audio transfer seems ok.

This is the python server:

C:\esp\pipeline_tcp_client\tools> C:\Python27\python.exe .\tcp_server.py
Get the esp32.mp3 size is 55302
starting listen on ip 192.168.1.69, port 8080
waiting for client to connect
total size ←[1;35m [1024/55302] ←[0m
total size ←[1;35m [1039/55302] ←[0m
total size ←[1;35m [1052/55302] ←[0m
total size ←[1;35m [1124/55302] ←[0m
total size ←[1;35m [1199/55302] ←[0m
total size ←[1;35m [1506/55302] ←[0m
total size ←[1;35m [1777/55302] ←[0m
total size ←[1;35m [2458/55302] ←[0m
total size ←[1;35m [2538/55302] ←[0m
total size ←[1;35m [2590/55302] ←[0m
total size ←[1;35m [3552/55302] ←[0m
total size ←[1;35m [3752/55302] ←[0m
total size ←[1;35m [4452/55302] ←[0m
total size ←[1;35m [4763/55302] ←[0m
total size ←[1;35m [5208/55302] ←[0m
total size ←[1;35m [5208/55302] ←[0m
get all data for esp32.mp3
 close client connect

This is the esp32 monitor:

I (29) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (29) boot: compile time 18:50:38
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00300000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20a0ch (133644) map
I (160) esp_image: segment 1: paddr=00030a34 vaddr=3ffb0000 size=04208h ( 16904) load
I (166) esp_image: segment 2: paddr=00034c44 vaddr=40080000 size=0b3d4h ( 46036) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9b74ch (636748) map
I (416) esp_image: segment 4: paddr=000db774 vaddr=4008b3d4 size=0b5a0h ( 46496) load
I (436) esp_image: segment 5: paddr=000e6d1c vaddr=50000000 size=00010h (    16) load
I (447) boot: Loaded app from partition at offset 0x10000
I (447) boot: Disabling RNG early entropy source...
I (458) cpu_start: Pro cpu up.
I (459) cpu_start: Starting app cpu, entry point is 0x400813c0
0x400813c0: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (473) cpu_start: Pro cpu start user code
I (473) cpu_start: cpu freq: 160000000
I (473) cpu_start: Application information:
I (477) cpu_start: Project name:     tcp_client_example
I (483) cpu_start: App version:      1
I (488) cpu_start: Compile time:     Jan  9 2022 18:50:18
I (494) cpu_start: ELF file SHA256:  654354a3668cb64f...
I (500) cpu_start: ESP-IDF:          v4.3.1
I (505) heap_init: Initializing. RAM available for dynamic allocation:
I (512) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (518) heap_init: At 3FFB80B8 len 00027F48 (159 KiB): DRAM
I (524) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (531) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (537) heap_init: At 40096974 len 0000968C (37 KiB): IRAM
I (544) spi_flash: detected chip: generic
I (548) spi_flash: flash io: dio
W (552) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (566) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (634) TCP_CLIENT_EXAMPLE: [ 1 ] Start codec chip
I (654) TCP_CLIENT_EXAMPLE: [2.0] Create audio pipeline for playback
I (654) TCP_CLIENT_EXAMPLE: [2.1] Create i2s stream to write data to codec chip
I (684) TCP_CLIENT_EXAMPLE: [2.2] Create mp3 decoder to decode mp3 file
I (694) TCP_CLIENT_EXAMPLE: [2.2] Create tcp client stream to read data
I (694) TCP_CLIENT_EXAMPLE: [2.3] Register all elements to audio pipeline
I (704) TCP_CLIENT_EXAMPLE: [2.4] Link it together tcp-->mp3-->i2s
I (704) TCP_CLIENT_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
E (714) gpio: gpio_install_isr_service(460): GPIO isr service already installed
W (2974) wifi:<ba-add>idx:0 (ifx:0, e8:d1:1b:05:99:55), tid:0, ssn:0, winSize:64
W (2994) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (4134) TCP_CLIENT_EXAMPLE: [ 4 ] Set up  event listener
I (4134) TCP_CLIENT_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (4134) TCP_CLIENT_EXAMPLE: [4.2] Listening event from peripherals
I (4144) TCP_CLIENT_EXAMPLE: [ 5 ] Start audio_pipeline
E (4214) AUDIO_ELEMENT: IN-[tcp] AEL_STATUS_ERROR_INPUT
E (4214) AUDIO_ELEMENT: [tcp] ERROR_PROCESS, AEL_IO_FAIL
W (4224) AUDIO_ELEMENT: [tcp] audio_element_on_cmd_error,3
I (4234) TCP_CLIENT_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=8000, bits=16, ch=2

The LyraT board, of course, is working fine with other demo.

The problem seems relatet to these errors:
E (4214) AUDIO_ELEMENT: IN-[tcp] AEL_STATUS_ERROR_INPUT
E (4214) AUDIO_ELEMENT: [tcp] ERROR_PROCESS, AEL_IO_FAIL

Any idea?
Thanks
Luca

shootao, 2022-01-18:
Hi @chitarrastonata 
According to the information you provided，the problem may be that the `tcp_server.py`(PC)  read file failed, then  `tcp_server.py` actively closes the connection.

chitarrastonata, 2022-01-18:
probably my English is not good enough, and i'm not able to explain my problem in a good view...

---

Python Server seems to load MP3 correctly. 
This is a message from Python Server:

C:\esp\pipeline_tcp_client\tools> C:\Python27\python.exe .\tcp_server.py
Get the esp32.mp3 size is 55302
starting listen on ip 192.168.1.69, port 8080
waiting for client to connect

---

Using a Windows TCP/IP test tools (hercules). I'm able to connect to the Python Server and get MP3 data.

---

The problem seems related to ESP32.
Esp32 starts to receive MP3, and it shows MP3 information:

I (4234) TCP_CLIENT_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=8000, bits=16, ch=2

but unfortunately no audio is played.

---

ESP32 before receiving the MP3 is showing two errors, that I suppose could be the problem:

E (4214) AUDIO_ELEMENT: IN-[tcp] AEL_STATUS_ERROR_INPUT
E (4214) AUDIO_ELEMENT: [tcp] ERROR_PROCESS, AEL_IO_FAIL

---

I hope this time it was more comprehensive 
Thanks
Luca

shootao, 2022-02-07:
Hi @chitarrastonata , Thank you for your detailed explanation
According to your log, the **tcp_server** actively disconnects after sending a little data, So there was an error `[tcp] ERROR_PROCESS, AEL_IO_FAIL`.
Now we have added some [debug information](https://github.com/espressif/esp-adf/blob/c5ef39d7c3ae3e3f4fb997b81dfc5d14b36ba299/components/audio_stream/tcp_client_stream.c#L112)

chitarrastonata, 2022-02-08:
Dear shootao,

I can confirm you that there is an error calling the function esp_transport_read()
Using your new source code an error message in now showing.

What is not clear to me is how to solve this problem.
I can also confirm that replacing the LyraT board with another TCP client application I receive all MP3 file without problem.

this is the new log from LyraT with the new source code:

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6924
ho 0 tail 12 room 4
load:0x40078000,len:14308
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (29) boot: compile time 18:50:38
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00300000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20a4ch (133708) map
I (160) esp_image: segment 1: paddr=00030a74 vaddr=3ffb0000 size=04208h ( 16904) load
I (167) esp_image: segment 2: paddr=00034c84 vaddr=40080000 size=0b394h ( 45972) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9b76ch (636780) map
I (416) esp_image: segment 4: paddr=000db794 vaddr=4008b394 size=0b5e0h ( 46560) load
I (436) esp_image: segment 5: paddr=000e6d7c vaddr=50000000 size=00010h (    16) load
I (447) boot: Loaded app from partition at offset 0x10000
I (447) boot: Disabling RNG early entropy source...
I (459) cpu_start: Pro cpu up.
I (459) cpu_start: Starting app cpu, entry point is 0x400813c0
0x400813c0: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (473) cpu_start: Pro cpu start user code
I (473) cpu_start: cpu freq: 160000000
I (473) cpu_start: Application information:
I (477) cpu_start: Project name:     tcp_client_example
I (483) cpu_start: App version:      1
I (488) cpu_start: Compile time:     Feb  7 2022 21:15:47
I (494) cpu_start: ELF file SHA256:  8101486bccdc7620...
I (500) cpu_start: ESP-IDF:          v4.3.1
I (505) heap_init: Initializing. RAM available for dynamic allocation:
I (512) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (518) heap_init: At 3FFB80B8 len 00027F48 (159 KiB): DRAM
I (524) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (531) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (537) heap_init: At 40096974 len 0000968C (37 KiB): IRAM
I (544) spi_flash: detected chip: generic
I (548) spi_flash: flash io: dio
W (552) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (566) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (634) TCP_CLIENT_EXAMPLE: [ 1 ] Start codec chip
I (654) TCP_CLIENT_EXAMPLE: [2.0] Create audio pipeline for playback
I (654) TCP_CLIENT_EXAMPLE: [2.1] Create i2s stream to write data to codec chip
I (694) TCP_CLIENT_EXAMPLE: [2.2] Create mp3 decoder to decode mp3 file
I (694) TCP_CLIENT_EXAMPLE: [2.2] Create tcp client stream to read data
I (694) TCP_CLIENT_EXAMPLE: [2.3] Register all elements to audio pipeline
I (704) TCP_CLIENT_EXAMPLE: [2.4] Link it together tcp-->mp3-->i2s
I (704) TCP_CLIENT_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
E (714) gpio: gpio_install_isr_service(460): GPIO isr service already installed
W (1614) wifi:<ba-add>idx:0 (ifx:0, a4:91:b1:c0:95:3d), tid:5, ssn:0, winSize:64
W (1614) wifi:<ba-add>idx:1 (ifx:0, a4:91:b1:c0:95:3d), tid:0, ssn:0, winSize:64
W (1644) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (2634) TCP_CLIENT_EXAMPLE: [ 4 ] Set up  event listener
I (2634) TCP_CLIENT_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (2634) TCP_CLIENT_EXAMPLE: [4.2] Listening event from peripherals
I (2644) TCP_CLIENT_EXAMPLE: [ 5 ] Start audio_pipeline
W (2714) TCP_STREAM: TCP server actively closes the connection
I (2714) TCP_CLIENT_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
W (2954) TCP_CLIENT_EXAMPLE: [ * ] Stop event received
I (2954) TCP_CLIENT_EXAMPLE: [ 6 ] Stop audio_pipeline
E (2954) AUDIO_ELEMENT: [tcp] Element already stopped
E (2964) AUDIO_ELEMENT: [mp3] Element already stopped
E (2964) AUDIO_ELEMENT: [i2s] Element already stopped
W (2974) AUDIO_PIPELINE: There are no listener registered
W (2974) AUDIO_ELEMENT: [tcp] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2984) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (2994) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3014) wifi:<ba-del>idx
W (3014) wifi:<ba-del>idx
W (3014) PERIPH_WIFI: Wi-Fi disconnected from SSID FASTWEB-C0953D, auto-reconnect disabled, reconnect after 1000 ms
W (3014) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:3

chitarrastonata, 2022-02-08:
Dear shootao,

 

I can confirm you that there is an error calling the function esp_transport_read()

Using your new source code an error message in now showing.

 

What is not clear to me is how to solve this problem.

I can also confirm that replacing the LyraT board with another TCP client application I receive all MP3 file without problem.

 

this is the new log from LyraT with the new source code:

 

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:6924

ho 0 tail 12 room 4

load:0x40078000,len:14308

load:0x40080400,len:3716

0x40080400: _init at ??:?

 

entry 0x40080680

I (29) boot: ESP-IDF v4.3.1 2nd stage bootloader

I (29) boot: compile time 18:50:38

I (29) boot: chip revision: 3

I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0

I (39) boot.esp32: SPI Speed      : 40MHz

I (44) boot.esp32: SPI Mode       : DIO

I (48) boot.esp32: SPI Flash Size : 4MB

I (53) boot: Enabling RNG early entropy source...

I (58) boot: Partition Table:

I (62) boot: ## Label            Usage          Type ST Offset   Length

I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000

I (76) boot:  1 phy_init         RF data          01 01 0000d000 00001000

I (84) boot:  2 factory          factory app      00 00 00010000 00300000

I (91) boot: End of partition table

I (96) boot_comm: chip revision: 3, min. application chip revision: 0

I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20a4ch (133708) map

I (160) esp_image: segment 1: paddr=00030a74 vaddr=3ffb0000 size=04208h ( 16904) load

I (167) esp_image: segment 2: paddr=00034c84 vaddr=40080000 size=0b394h ( 45972) load

I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9b76ch (636780) map

I (416) esp_image: segment 4: paddr=000db794 vaddr=4008b394 size=0b5e0h ( 46560) load

I (436) esp_image: segment 5: paddr=000e6d7c vaddr=50000000 size=00010h (    16) load

I (447) boot: Loaded app from partition at offset 0x10000

I (447) boot: Disabling RNG early entropy source...

I (459) cpu_start: Pro cpu up.

I (459) cpu_start: Starting app cpu, entry point is 0x400813c0

0x400813c0: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:141

 

I (0) cpu_start: App cpu up.

I (473) cpu_start: Pro cpu start user code

I (473) cpu_start: cpu freq: 160000000

I (473) cpu_start: Application information:

I (477) cpu_start: Project name:     tcp_client_example

I (483) cpu_start: App version:      1

I (488) cpu_start: Compile time:     Feb  7 2022 21:15:47

I (494) cpu_start: ELF file SHA256:  8101486bccdc7620...

I (500) cpu_start: ESP-IDF:          v4.3.1

I (505) heap_init: Initializing. RAM available for dynamic allocation:

I (512) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM

I (518) heap_init: At 3FFB80B8 len 00027F48 (159 KiB): DRAM

I (524) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM

I (531) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM

I (537) heap_init: At 40096974 len 0000968C (37 KiB): IRAM

I (544) spi_flash: detected chip: generic

I (548) spi_flash: flash io: dio

W (552) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.

I (566) cpu_start: Starting scheduler on PRO CPU.

I (0) cpu_start: Starting scheduler on APP CPU.

I (634) TCP_CLIENT_EXAMPLE: [ 1 ] Start codec chip

I (654) TCP_CLIENT_EXAMPLE: [2.0] Create audio pipeline for playback

I (654) TCP_CLIENT_EXAMPLE: [2.1] Create i2s stream to write data to codec chip

I (694) TCP_CLIENT_EXAMPLE: [2.2] Create mp3 decoder to decode mp3 file

I (694) TCP_CLIENT_EXAMPLE: [2.2] Create tcp client stream to read data

I (694) TCP_CLIENT_EXAMPLE: [2.3] Register all elements to audio pipeline

I (704) TCP_CLIENT_EXAMPLE: [2.4] Link it together tcp-->mp3-->i2s

I (704) TCP_CLIENT_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network

E (714) gpio: gpio_install_isr_service(460): GPIO isr service already installed

W (1614) wifi:<ba-add>idx:0 (ifx:0, a4:91:b1:c0:95:3d), tid:5, ssn:0, winSize:64

W (1614) wifi:<ba-add>idx:1 (ifx:0, a4:91:b1:c0:95:3d), tid:0, ssn:0, winSize:64

W (1644) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4

I (2634) TCP_CLIENT_EXAMPLE: [ 4 ] Set up  event listener

I (2634) TCP_CLIENT_EXAMPLE: [4.1] Listening event from all elements of pipeline

I (2634) TCP_CLIENT_EXAMPLE: [4.2] Listening event from peripherals

I (2644) TCP_CLIENT_EXAMPLE: [ 5 ] Start audio_pipeline

W (2714) TCP_STREAM: TCP server actively closes the connection

I (2714) TCP_CLIENT_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2

W (2954) TCP_CLIENT_EXAMPLE: [ * ] Stop event received

I (2954) TCP_CLIENT_EXAMPLE: [ 6 ] Stop audio_pipeline

E (2954) AUDIO_ELEMENT: [tcp] Element already stopped

E (2964) AUDIO_ELEMENT: [mp3] Element already stopped

E (2964) AUDIO_ELEMENT: [i2s] Element already stopped

W (2974) AUDIO_PIPELINE: There are no listener registered

W (2974) AUDIO_ELEMENT: [tcp] Element has not create when AUDIO_ELEMENT_TERMINATE

W (2984) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE

W (2994) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE

W (3014) wifi:<ba-del>idx

W (3014) wifi:<ba-del>idx

W (3014) PERIPH_WIFI: Wi-Fi disconnected from SSID FASTWEB-C0953D, auto-reconnect disabled, reconnect after 1000 ms

W (3014) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:3

A: espressif/esp-adf
Cc: Luca Moreschi; Mention

chitarrastonata, 2022-02-08:
****** SOLVED ******
Dear shootao,

On my machine python server implementation does not works fine.
Python server start to send data, but instead to send 1024 byte for each block, send small block and close connection.

So the problem is not in your ESP32 library, but in the server implementation:
As already send in previous message, this is the python server log:

Get the esp32.mp3 size is 55302
starting listen on ip 192.168.1.69, port 8080
waiting for client to connect
total size ←[1;35m [1024/55302] ←[0m
total size ←[1;35m [1039/55302] ←[0m
total size ←[1;35m [1052/55302] ←[0m
total size ←[1;35m [1124/55302] ←[0m
total size ←[1;35m [1199/55302] ←[0m
total size ←[1;35m [1506/55302] ←[0m
total size ←[1;35m [1777/55302] ←[0m
total size ←[1;35m [2458/55302] ←[0m
total size ←[1;35m [2538/55302] ←[0m
total size ←[1;35m [2590/55302] ←[0m
total size ←[1;35m [3552/55302] ←[0m
total size ←[1;35m [3752/55302] ←[0m
total size ←[1;35m [4452/55302] ←[0m
total size ←[1;35m [4763/55302] ←[0m
total size ←[1;35m [5208/55302] ←[0m
total size ←[1;35m [5208/55302] ←[0m
get all data for esp32.mp3
 close client connect

First block is sent correctly with 1024 byte [1024/55302],
but starting from the second block send less byte, second block only 15 byte [1039/55302], and so on....
last block is only at [5208/55302], and then close connection. 

I wrote on my own a C/C++ server implementation in Visual Studio 2019,
and now everything is working fine! 

Let me know if someone need my C/C++ server....

Luca Moreschi (chitarrastonata)

chitarrastonata, 2022-02-08:
Dear shootao,

I solved it!

It was the python server problem.

I already have update on GitHub with the details.

 

Now I need to implement a similar application using UDP instead TCP.

I suppose I have to implement it by myself.

There is some documentation or support to implement a new protocol?

UDP and TCP are very similar so, I should not be so difficult to do it.

Thanks

Luca Moreschi (chitarrastonata)

closed 2022-02-09 by jason-mao