espressif/esp-adf #744: example pipeline_recording_to_sdcard cannot record sound using LyraT-Mini v1.2

jw-redpanda, 2022-01-10:
Referring to the example pipeline_recording_to_sdcard ([link](https://github.com/espressif/esp-adf/tree/master/examples/recorder/pipeline_recording_to_sdcard)), when using with LyraT-Mini v1.2, the recorded opus file has no sound and is silent.  Already applied idf.py menuconfig and select "Audio HAL > Audio board (ESP32-Lyrat-Mini V1.1)"

One warning message is seen below:
`W (1667) AUIDO_MEM: Can't support stack on external memory due to ESP32 chip is 1`

When the same project runs on LyraT v4.3 (menuconfig to select "ESP32-LyraT V4.3"), it can record sound as expected without issue and no AUDIO_MEM warning is seen.

Using the versions below:
```
esp-idf: ESP-IDF v4.4-beta1-284-gd83021a6e8-dirty or ESP-IDF v4.1.1-44-g83aaee86d-dirty
esp-adf: bd305a6dd5e96f019e913d0ff3f407192b18b084 (latest as of 2022-01-10)
freertos patch: applied idf_v4.4_freertos.patch

```

I guess some tweaks may be required for the LyraT Mini V1.2 board.  
Any hints?

nezen, 2022-01-27:
I tested this example on my lyrat-mini, it turned out it a wav file can be recorded, but when I tried to record it using opus format, an opus file can be generated but none of my players(vlc, foobar, audacity) were able to recognize the file.

I was using:
IDF4.4 f3e0c8bc41098f79baf2cf9f45dcbb6c1d41b3fe
latest ADF c5ef39d

jason-mao, 2022-02-07:
> Referring to the example pipeline_recording_to_sdcard ([link](https://github.com/espressif/esp-adf/tree/master/examples/recorder/pipeline_recording_to_sdcard)), when using with LyraT-Mini v1.2, the [...]

The opus encoder need more RAM than others,  you can disable the `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` to clear the error message.

rylox, 2022-03-06:
> I tested this example on my lyrat-mini, it turned out it a wav file can be recorded, but when I tried to record it using opus format, an opus file can be generated but none of my players(vlc, [...]

I'm having issues to get the wav example working on my LyraT mini, it's v1.2 aswell. The recorded wav file varies in length each test, has very poor quality and the audio is speed up each second, about 4x times speed at the end. Did you do anything special like enabling some kind of feature or did you just flash the plain example with the menuconfig for the right board and codec?

jason-mao, 2022-08-09:
@rylox Please test the latest code. We found a related issue may be resolved this question.

I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2022-08-09 by jason-mao